wled/WLED #2944: mDNS address is saved but not used

emmettprexus, 2022-12-11:
### What happened?

I specified "wled-entrance" as my mDNS address but it is inaccessible eventhough the ESP32 is connected to wifi. I found out via my dhcp table in the router that it uses wled-WLED as mDNS address instead.

### To Reproduce Bug

Set an mDNS address to "merry-christmas" and let it connect to wifi. `ping merry-christmas` and you will get no responses. `ping wled-WLED` and you'll see that that is the mDNS address it chose to go with.

### Expected Behavior

I expected it to use the mDNS address that i provided

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b0 (build 2212050)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-11:
Did you re-check WiFi settings page to verify it is saved properly?
I cannot reproduce your findings.
[image]
[image]

Aircoookie, 2022-12-11:
Routers will typically show the _hostname_ in the DHCP table, not the mDNS name.
The hostname is based off the "Server Description" in UI settings.
You can test on a supported device (most phones are OK, on Windows you'll need to install Apple Bonjour) whether `merry-christmas.local` is accessible in a web browser.

emmettprexus, 2022-12-12:
> Did you re-check WiFi settings page to verify it is saved properly? I cannot reproduce your findings. [image] [image]

Yes, the settings was correctly saved and persisted fine during reboots.

> Routers will typically show the _hostname_ in the DHCP table, not the mDNS name. The hostname is based off the "Server Description" in UI settings. You can test on a supported device (most phones [...]

I'm only able to ping wled-WLED/wled-WLED.local and not merry-christmas.local and same goes for access through the webinterface, only http://wled-WLED responds, and http://merry-christmas.local is dead. 

Other esp32 driven stuff I have on my network works just fine using their mDNS addresses and also shows up in my dhcp table with their mDNS name. 

Strange...

emmettprexus, 2022-12-12:
I changed to a ESP32-PICO and installed v0.13.3 via the website and now it responds to merry-christmas. The other esp32 died over night so maybe it was a mondays edition or something was off with my own binary🤷‍♂️ At least it's responding correctly to mDNS pings now and it shows up correctly in unifi's device list after updating the hostname under the UI settings.

Thank you for wled! :)

closed 2022-12-12 by emmettprexus