wled/WLED #2945: Unable to set segment name in 2D mode since 0.14.0

simonemarin, 2022-12-11:
### What happened?

I am not able to set a segment name anymore after moving to 0.14.0

### To Reproduce Bug

- Set 2D mode in configuration

- In the web GUI click in the pencil icon and set the text field

- Use the check mark (I guess that is like the APPLY button in 0.13.3 ?)

- Name is not applied

### Expected Behavior

I expect the name to be saved after clicking the check mark icon

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.0-b0 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

I can properly change the name of the segment with the JSON API both in a preset and with external API http POST request

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

simonemarin, 2022-12-12:
[image]

I had a look at the html and JS console to see if I can spot something weird but it seems ok to me, no errors

blazoncek, 2022-12-12:
Tested on ESP32 and ESP8266 and cannot reproduce.
You may need to specify test conditions in more detail and/or provide cfg.json and presets.json.

gamer1966, 2022-12-13:
Tested on different ESP32 (Dev-Kit-C, S3, ESP32 NodeMCU D1). I can reproduce the error. Furthermore no changes are saved.

[image]

Simple setup
[image]

Before
[image]

Edit
[image]

After pressing the checkmark
[image]

softhack007, 2022-12-13:
Looks like you also modified the "StopY" and "StopX" fields, am I correct?

Maybe the exact sequence of that you did is needed to make the problem reproducible.
Did you first change X/Y values, or first changed the name then modified X/Y?

gamer1966, 2022-12-13:
Error must be in 43582b6. Before this everything works fine.

[image]

softhack007, 2022-12-13:
> If I leave "html_ui.h" unchanged and only use the changes in "index.js" and [..] everything works fine.
> I don't know what "html_ui.h" is used for.

The way you tried, the change does not have _any_ effect. html_ui.h (and other html files) are generated from the files in data/, like index.js. After changing index.js, you need to run "npm run build" for the changes to take effect.
My suggestion is actually to wait for one of the UI developers (@blazoncek or @ewoudwijma ?) to analyze the problem, and they might come up with a proper solution.

Please also answer the other question: Did you also manually edit  "StopY" and "StopX" fields? If yes, did you edit the preset name before or after?

gamer1966, 2022-12-13:
> The way you tried, the change does not have _any_ effect. html_ui.h (and other html files) are generated from the files in data/, like index.js. After changing index.js, you need to run "npm run [...]

Well, i use Visual Studio Code to build. So I don't run "npm run build". 
First I change the segment name then I change Stop X and Stop Y. After this I press the checkmark.

softhack007, 2022-12-13:
Thanks, this detail may be important.

> Well, i use Visual Studio Code to build. So I don't run "npm run build".

See https://kno.wled.ge/advanced/custom-features/#changing-web-ui

blazoncek, 2022-12-13:
BTW your panel setup seems out of place as it is 1D

gamer1966, 2022-12-13:
> Thanks, this detail may be important.
> 
> If you do it in two steps - edit, change Stop X and Stop Y, press checkmark, edit, change name - does it work better?
> 
> > Well, i use Visual Studio Code to build. So I don't run "npm run build".
> 
> See https://kno.wled.ge/advanced/custom-features/#changing-web-ui

I'm a little bit confused. First I instaled node.js and after a reboot i run "npm run build".
After this it works correct. Then I change `stopY>eY.max` back to `stop>eY.max` and the old error is back.
Now I change `stop>eY.max` to `stopY>eY.max` back and the error is not corrected.
I run "npm run build" multible time but I can not save the changes anymore.

gamer1966, 2022-12-13:
[image]

Thank You, this works.

closed 2022-12-13 by blazoncek