wled/WLED #2950: Sunset time is not calculating correctly

TeslaBenny, 2022-12-14:
### What happened?

The Sunset time is not calculating and showing as 00:00.
[image]

### To Reproduce Bug

Open Config > Time & Macros.
Set local NTP server, timezone, latitude, and longitude settings.
Save the settings.

### Expected Behavior

Sunset should show a valid time based on configured timezone, latitude, and longitude values.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-15:
This may occasionally happen (known issue) due to rounding errors. It will go away next day.

closed 2022-12-15 by blazoncek

TeslaBenny, 2022-12-15:
Thank you @blazoncek! Where is the known issue documented? I wasn't able to find it.

If the rounding errors aren't fixed yet, may I take a stab at the logic and attempt to provide a resolution to the issue? I just need to know which files and lines of code are involved with the sunset (and sunrise too?) calculation.

blazoncek, 2022-12-15:
Just check closed issues.
No need to check the code as it is intentional that way (using custom math functions which favor code size over accuracy).
You can compile using -D WLED_USE_REAL_MATH to use double precision functios.

jmerc77, 2023-04-22:
this just happened to me on 4/21/23. should have been 20:01 for me

blazoncek, 2023-04-22:
> this just happened to me on 4/21/23. should have been 20:01 for me. using esp32 s3.

And it went away on the next day. Are you using 0.14?

jmerc77, 2023-04-22:
0.14-b1 to be exact. day not over yet but assume it is temporary as this is the first time i noticed. may find out in about 2.5h.