wled/WLED #2952: Backing up presets via presets.json

githubsteveb, 2022-12-15:
### What happened?

Hi there, reading [Backing-up and Restoring Presets](https://kno.wled.ge/features/presets/#backing-uprestoring-presets) it say to do the following:

> To backup all presets, go to [WLED-IP]/edit (OTA lock must be off), right click the presets.json file and download!
To restore, use the Choose file and Upload buttons.

However, the presets.json file is empty:
[image]

Is this a bug ?

Thanks for a wicked piece of software!

### To Reproduce Bug

> To backup all presets, go to [WLED-IP]/edit (OTA lock must be off), right click the presets.json file and download!

### Expected Behavior

presets.json is populated with the existing saved preset.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-15:
The info in KB is a bit outdated (still valid though).
Use backup & restore buttons in Security settings.

githubsteveb, 2022-12-15:
> The info in KB is a bit outdated (still valid though). Use backup & restore buttons in Security settings.

Thanks - I didn't know about that!

And I've released my original error - there is a massive amount of white space (which I didn't realise ) so I need to right scroll and all the preset config is there! 😆
[image]

closed 2022-12-15 by githubsteveb