wled/WLED #2956: Only 1 of 2 APA102 strips on 13.3 comes back up after a power cycle

nwalke, 2022-12-15:
### What happened?

I power cycled wled and when it came back, only 1 of my 2 APA102 strips was configured.  I had to remove both and add them back in to settings before both came back up like they were previous to the power cycle.

### To Reproduce Bug

Configure 2 APA102 strips (my lengths are 557 and 257, not sure if that matters), power cycle WLED, observe that only one comes back up.

### Expected Behavior

Both strips configured in the settings would come back online after a WLED power cycle.

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

The two strips still show as configured in the settings even though only one works after a power cycle.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

closed 2022-12-15 by blazoncek

nwalke, 2022-12-15:
@blazoncek  Is there somewhere I can read more about that limitation?  I can get both working, just requires some settings page wonkiness.

blazoncek, 2022-12-16:
APA102 uses SPI interface. NeoPixelBus only supports (AFAIK) only one LED strip on SPI. WLED only uses HSPI on ESP32, VSPI is reserved for other use.