wled/WLED #2957: Loxone integration not working with 13.3

dkennerknecht, 2022-12-16:
### What happened?

With the upgrade from **WLED 0.12.0** (build 2104020) to **WLED 0.13.3** (build 2208222) 
the integration into Loxone is no longer working. 

lx | BBBGGGRRR: 0 - 100100100 | Loxone RGB value for primary color. Each color (RRR,GGG,BBB) is specified in the range from 0 to 100%.
-- | -- | --

I have also tried the version with `20bbbtttt` but it take no effect.

I also tried the v13.0, v13.1 and v13.2 but i looks like that the Loxone parameters aren't working since v13.0
I already checked the code and `WLED_ENABLE_LOXONE` is per default enabled

### To Reproduce Bug

The following command will work wit 12.0  
`curl -X POST "http://<IP_ADDRESS>/json" -d '{"seg":{"lx":100100100}' -H "Content-Type: application/json"`
When u try it with any v13.x version nothing will happen but you get the response `"success":true`

### Expected Behavior

The Loxone Parser should set the correct RGB(W) values 

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-16:
Try 0.14 and report.

Clubsport, 2022-12-16:
Cannot confirm that.

13.3 (without sound reactive) works fine for me on ESP32.

dkennerknecht, 2022-12-16:
> Try 0.14 and report.

Is v0.14 already released and where can i find it?

softhack007, 2022-12-16:
You can find the beta of 0.14 here
* https://wled-install.github.io/
* https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield/v0.14.0-b0

dkennerknecht, 2022-12-17:
@softhack007 @blazoncek 
With v0.14, it's also not working.

softhack007, 2022-12-17:
@blazoncek, according to git blame, the line below was modified 5 month ago, while the rest from lx_parser.cpp is the same since two years. Could it be related to the problem that @dkennerknecht sees ?

https://github.com/Aircoookie/WLED/blob/3c5838cafd77b75b336166afdeef7ff52556bea0/wled00/lx_parser.cpp#L72

blazoncek, 2022-12-17:
@dkennerknecht Can you confirm, that you have binary with Loxone compiled in?

@softhack007 that has nothing to do with ability to change color. 0.14 changed API.

blazoncek, 2022-12-17:
Due to API changes this may be more efficient and semantically correct.
[image]

blazoncek, 2022-12-17:
I have tested the above code `{"seg":{"lx":100100100}}` and my LEDs turned white. I do not know if this is ok or not. but it seems to me that Loxone is working.
This is what I get on debug output:
```
JSON buffer locked. (14)
LX: Lox primary = 100100100
LX: Lox = 100100100
LX: turn on
LX: segment 0
LX: Lox primary = 100100100
LX: Lox = 100100100
LX: segment 1
JSON buffer released. (14)
JSON buffer locked. (12)
```

blazoncek, 2022-12-17:
It does the same with original source and the modified in a post above.

dkennerknecht, 2022-12-17:
@blazoncek The missing end bracket was a typo.
As binary im always using the uploaded versions on git and for those the LOXONE parser should be enabled as default.

`lx: BBB GGG RRR` when 100100100 is used it should be white, (000000)100 will be red 100%.

@blazoncek It's not so easy to enable the debug an connect via Serial ... for that i have to build a test setup but lets try it.

blazoncek, 2022-12-18:
Please check if e8a7802e946028403b97810b3365e0d83c05e520 solves it.
If it does, please close the issue.

dkennerknecht, 2022-12-18:
@blazoncek your last commit https://github.com/Aircoookie/WLED/commit/e8a7802e946028403b97810b3365e0d83c05e520 solved the issue
Thank you

closed 2022-12-18 by dkennerknecht