wled/WLED #2961: Glitching

TysTies, 2022-12-19:
### What happened?

I have weld installed on lots of devices. esp8266 & esp32. All of them do this except one. When I turn them on it will start the default preset, I set and like pause for half a seconded. Then resume and then do it again in about ten seconds. It's completely random. I wonder if it is like a cooling issue? What is proper cooling? 

### To Reproduce Bug

 n/a

### Expected Behavior

n/a

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build2208222)

### Which microcontroller/board are you seeing the problem on?

Other

### Relevant log/trace output

```shell
n/a
```

### Anything else?

n/a

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-19:
Do you have Home Assistant integration?

TysTies, 2022-12-19:
I don't

TysTies, 2022-12-19:
I also had a esp32 that ran the fire effect on 7 different segments overnight and my board wont work any more

TysTies, 2022-12-20:
https://photos.app.goo.gl/5pgitKQdUitUwXX38 a video.

blazoncek, 2022-12-20:
A little less camera movement and shorter video with exact symptom would be more helpful.

blazoncek, 2022-12-20:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
You may need level shifter in your set-up.
And you will need a debug build to diagnose such issues.

TysTies, 2022-12-20:
wdm by debug build?

Aircoookie, 2022-12-21:
Sounds like an unsucessful network request is blocking the main loop. Ensure the module has internet access, alternatively make sure to disable NTP in time settings and check that MQTT/Blynk/Hue integrations are disabled.

TysTies, 2022-12-21:
I'm not using the Wi-Fi; I use hotspot data @ my house. (It is cheaper in my area) I just control them via the connect to network interface. Would having the power limiter tuned off and insufficient power be a possible cause? I replaced the broken board and the same thing happened within minutes (I'm stupid enough to turn that off and leave it during setup)

TysTies, 2022-12-23:
That Hypostasis was correct, When the Power is insufficient the boards Wi-Fi is ruined.

closed 2022-12-23 by TysTies