wled/WLED #2962: Lost segments after updating from 0.13.0-b6 to 0.13.3 on ESP8266 trough home assistant

Megachip, 2022-12-19:
### What happened?

After upgrade through HA, it seems that all segments are lost. All other configurations stayed (like presents).

### To Reproduce Bug

Hit the upgrade button on home assistant.

### Expected Behavior

no data/config lost.

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2022-12-19:
Segments are part of a preset in newer versions. If you activate a preset, does that change segments?

Megachip, 2022-12-20:
> Segments are part of a preset in newer versions. If you activate a preset, does that change segments?

Will check asap. But afair all presets with segments didn't work at all anymore. Also HomeAssistant reporting all segments as missing (but didn't know, how the communication with HA works now, if segments are part of the presets)

blazoncek, 2022-12-20:
Segments are temporary at best. They can be stored in presets if you select to save segment bounds.
It is mandatory to save segment bounds in preset if you want to keep segments between reboots.

closed 2024-04-23 by github-actions[bot]