wled/WLED #2971: Sharing my awesome christmass tree effects, integration to wled

MatejsLab, 2022-12-24:
**Is your feature request related to a problem?** I would like to share some pleasant, casual effects for Christmas lights. I feel like they are missing in wled. My Code can be run as is on Arduino boars with FastLed library, but not yet on wled.

**Describe the solution you'd like** I have some cool code that is running unique effects on my Christmas lights, that wled comunity may enjoy, the total runtime of effects is about 10 minutes, so no one gets bored looking at lights, even i after 4 years.

The effects are based on randomized colors, and effects like half-interchangeable flashing, dimming, running lights, and more. Only 2 of them like dimming colors exist in wled.

**Describe alternatives you've considered**
My goal is to share ideas at least, the code is not optimal for ESP as it is running on delays, but the effects may be copied over into wled code and made publicly available, tho i don't know how to make it smoothly run for wled, nor have time to mess with it just yet. 

Another wled improvement wish is effect grouping in wled to separate flashy from, casual. Now it is too many options in one place, and hard to achieve similar effects i managed to make with some code years ago, mainly casual slow, vivid effects, and nonrepeatable in the last 10 minutes. Anyone older hates flashy effects but is amazed by non-repeatable ones.

Disclaimer: the code may heve bugs, or not work as expected, but it worked for me in 2018 I compiled it for my Christmas tree.
I may only fix it if i find it is not working for me.

```

#include <FastLED.h>

 #define NUM_LEDS 98
 #define DATA_PIN_L 3
 
 int t = 10;
int T =50;
int b;
int c;
int G,G2;
int R,R2;
int B,B2;
int RD;
int MO;
int ZE;
int Gp;
int Rp;
int Bp;
int n;
int p;
int te;
int rn;
byte pol,cik,cak;

CRGB leds[NUM_LEDS];

void setup() 
{
       FastLED.addLeds<WS2811, DATA_PIN_L, RGB>(leds, NUM_LEDS);
       }
void loop()
{
   //CIK CAK - interchange- half flash
  for(te=0;te<=16;te++)
  {
      R=random(0,4)*63;
      G=random(0,4)*63;
      B=random(0,4)*63;
   
	for(b=0;b<NUM_LEDS;b=b+2){ leds[b] = CRGB(R,G,B);}
        FastLED.show(); 
	delay(2000); 
        for(b=0;b<NUM_LEDS;b=b+2){ leds[b] = CRGB(0,0,0);}

	for(c=1;c<NUM_LEDS;c=c+2){ leds[c] = CRGB(R,G,B);}
        FastLED.show();
	delay(2000); 
        for(c=1;c<NUM_LEDS;c=c+2){ leds[c] = CRGB(0,0,0);}
  }  
  
  //NAKLJUCNI PRELIV   random dimming
   for(te=0;te<=16;te++)
  {
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
	
  for(n=0;n<1024;n++)
    {
	if(R!=RD)
	  {
	    if(Rp<R) RD++;
	    else RD--;
	  }			
			
	if(G!=ZE)
	  {
	   if(Gp<G) ZE++;
	   else ZE--;
	  }
				
	if(B!=MO)
	  {			
	   if(Bp<B)MO++;
	   else MO--;
	  }
    b=random(0,NUM_LEDS);								
   leds[b] = CRGB(RD,ZE,MO); 
           FastLED.show();
         delay(t);					
	}
  }
  
  //PRIZGE 2xPOL  UGASNE 2xPOL   interchange- double flash
  for(te=0;te<=16;te++)
  {
    pol=!pol;
    if(pol==1)
    {
      R=random(0,4)*63;
      G=random(0,4)*63;
      B=random(0,4)*63;
    }
    else
    {
      R=0;
      G=0;
      B=0;
    }
	for(b=0;b<NUM_LEDS;b=b+2)
	{ 
           leds[b] = CRGB(R,G,B);
	}
         FastLED.show(); 
	delay(1000); 

 if(pol==0)
    {
      R=random(0,4)*63;
      G=random(0,4)*63;
      B=random(0,4)*63;
    }
    else
    {
      R=0;
      G=0;
      B=0;
    }
	for(c=1;c<NUM_LEDS;c=c+2)
	{
          leds[c] = CRGB(R,G,B);	
	}
        FastLED.show();
	delay(1000); 

  }  
  
   
  //TEĆENJE  running colors

  for(te=0;te<=16;te++)
  {
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
	for(b=0;b<NUM_LEDS;b++)
	{ 
           leds[b] = CRGB(R,G,B);
           FastLED.show();  		
	   delay(t*5); 
	}
	delay(T); 

  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
	for(c=NUM_LEDS-1;c>=0;c--)
	{
          leds[c] = CRGB(R,G,B);
          FastLED.show();	
	  delay(t*5);	
	}
	delay(T); 

  }  
  
	//"PRELIVANJE" dimming colors
for(p=0;p<=16;p++)
{
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
	
  for(n=0;n<256;n++)
    {
	if(R!=RD)
	  {
	    if(Rp<R) RD++;
	    else RD--;
	  }			
			
	if(G!=ZE)
	  {
	   if(Gp<G) ZE++;
	   else ZE--;
	  }
				
	if(B!=MO)
	  {			
	   if(Bp<B)MO++;
	   else MO--;
	  }
									
   for(b=0;b<NUM_LEDS;b++)leds[b] = CRGB(RD,ZE,MO); 
           FastLED.show();
         delay(t);  					
	}
  Rp=R;
  Gp=G;
  Bp=B;
}

// KRIZNO TECENJE cross running colors
for(te=0;te<=16;te++)
  {
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
    
  R2=random(0,4)*63;
  G2=random(0,4)*63;
  B2=random(0,4)*63;
    for(n=0;n<NUM_LEDS;n++)
      {
         leds[b] = CRGB(R,G,B);
         FastLED.show();
         leds[c] = CRGB(R2,G2,B2);
         FastLED.show();
	 delay(t*5);	
       
         b++;
         c--;
      }
      b=0;
      c=NUM_LEDS;
      delay(T);
  } 
}
```

blazoncek, 2022-12-24:
Maš kašen video tega?

TysTies, 2022-12-24:
I uploaded this to an Arduino, and I love it! Please add this to WLED!

blazoncek, 2022-12-24:
One thing of note: WLED effects have no knowledge of the past or the next iteration. Each frame exists only for the time being displayed.
Constructs like `delay(2000)` are not allowed as each frame is calculated approximately 42 times per second.

softhack007, 2022-12-24:
@MatejsLab could you add some more comments to the code, and maybe use good practice of "verbose variable naming"? 

Because right now it's almost impossible to understand what your code does (for example what is `if(B!=MO)` for). 

It's really problematic atm to understand the logic of your effects. Can you share a video about how it looks like?

In WLED your single effect would be around 6 different effects, running in a playlist that takes care of timing. Delay cannot be used, however there are means to "delay" an effect for a certain time. But the necessary global state information needs to be reduced for sure.

MatejsLab, 2022-12-26:
Here is a version with some comments, I hope it is more logical now even tho it's WO descriptive variables yet, causes multi-rename conflicts with other code sadly, so I didn't feel like fixing it by hand yet.

I will add video soon.

```
#include <FastLED.h>

 #define NUM_LEDS 98
 #define DATA_PIN_L 3
 
 int t = 10; // pattern time
int T = 50; // waiting time
int b;      // odd LEDs array counter
int c;      // even LEDs array counter
int G,G2;   // intensity odd green, even green
int R,R2;   // intensity odd red, even red
int B,B2;   // intensity odd blue, even blue
int RD;     // LED output red
int MO;     // LED output blue
int ZE;     // LED output green
int Gp;     // Last state green
int Rp;     // Last state red
int Bp;     // Last state blue
int n;      // 3rd array counter (randomizer)
int p;      // 4th array counter (dimming colors)
int te;     // 5th array counter (running LEDs)

byte pol; // odd/even state holder

CRGB leds[NUM_LEDS];

void setup() 
{
       FastLED.addLeds<WS2811, DATA_PIN_L, RGB>(leds, NUM_LEDS);
       }
void loop()
{
   //CIK CAK - interchange- half flashing
  for(te=0;te<=16;te++)  // repeat pattern 16 times
  {
      R=random(0,4)*63; // generate colors
      G=random(0,4)*63;
      B=random(0,4)*63;
   
	for(b=0;b<NUM_LEDS;b=b+2){ leds[b] = CRGB(R,G,B);} // write colors odd leds, than show all
        FastLED.show(); 
	delay(2000); 
        for(b=0;b<NUM_LEDS;b=b+2){ leds[b] = CRGB(0,0,0);} // OFF odd leds, than show all

	for(c=1;c<NUM_LEDS;c=c+2){ leds[c] = CRGB(R,G,B);} // write colors even leds, than show all
        FastLED.show();
	delay(2000); 
        for(c=1;c<NUM_LEDS;c=c+2){ leds[c] = CRGB(0,0,0);}  // OFF even leds, than show all
  }  
  
  //NAKLJUCNI PRELIV - random color dimming
   for(te=0;te<=16;te++) // repeat pattern 16 times
  {
  R=random(0,4)*63; // generate color
  G=random(0,4)*63;
  B=random(0,4)*63;
	
  for(n=0;n<1024;n++) // 1023 times itteration  
    {
	if(R!=RD) // slowly adjust red 
	  {
	    if(Rp<R) RD++; 
	    else RD--;
	  }			
			
	if(G!=ZE) //slowly adjust green 
	  {
	   if(Gp<G) ZE++;
	   else ZE--;
	  }
				
	if(B!=MO) //slowly adjust blue 
	  {			
	   if(Bp<B)MO++;
	   else MO--;
	  }
    b=random(0, NUM_LEDS);		// randomly fill in a new color - light by light 		 				
   leds[b] = CRGB(RD,ZE,MO); 
           FastLED.show();
         delay(t);					
	}
  }
  
  //PRIZGE 2xPOL  UGASNE 2xPOL  //interchange- double flash
  for(te=0;te<=16;te++) // repeat patern 16 times
  {
    pol=!pol; // half by half state change
    if(pol==1) // odd leds active
    {
      R=random(0,4)*63; // set colors
      G=random(0,4)*63;
      B=random(0,4)*63;
    }
    else
    {
      R=0; // set OFF
      G=0;
      B=0;
    }
	for(b=0;b<NUM_LEDS;b=b+2) // fill in odd leds array and show
	{ 
           leds[b] = CRGB(R,G,B);
	}
         FastLED.show(); 
	delay(1000); 

 if(pol==0) // even leds active
    {
      R=random(0,4)*63; // set colors
      G=random(0,4)*63; 
      B=random(0,4)*63;
    }
    else
    {
      R=0; // set OFF
      G=0;
      B=0;
    }
	for(c=1;c<NUM_LEDS;c=c+2) // fill in even leds array and show
	{ 
          leds[c] = CRGB(R,G,B);	
	}
        FastLED.show();
	delay(1000); 

  }  
  
   
  //TEĆENJE running colors

  for(te=0;te<=16;te++) // repeat patern 16 times
  {
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
  
	for(b=0;b<NUM_LEDS;b++) // running up - one by one LED, set new color and show
	{ 
           leds[b] = CRGB(R,G,B);
           FastLED.show();  		
	   delay(t*5); 
	}
	delay(T); 

  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
  
	for(c=NUM_LEDS-1;c>=0;c--) // running down - one by one LED, set and show
	{
          leds[c] = CRGB(R,G,B);
          FastLED.show();	
	  delay(t*5);	
	}
	delay(T); 

  }  
  
	//"PRELIVANJE" dimming colors
for(p=0;p<=16;p++)
{
  R=random(0,4)*63;
  G=random(0,4)*63;
  B=random(0,4)*63;
	
  for(n=0;n<256;n++) // 255 available iterations to match new color of full array
    {
	if(R!=RD) // slowly match red
	  {
	    if(Rp<R) RD++;
	    else RD--;
	  }			
			
	if(G!=ZE) // slowly match green
	  {
	   if(Gp<G) ZE++;
	   else ZE--;
	  }
				
	if(B!=MO) //slowly match blue
	  {			
	   if(Bp<B)MO++;
	   else MO--;
	  }
									
   for(b=0;b<NUM_LEDS;b++)leds[b] = CRGB(RD,ZE,MO); // fill full led array and than show each iteration
           FastLED.show();
         delay(t);  					
	}
  Rp=R; // save last red
  Gp=G; // save last green
  Bp=B; // save last blue
}

// KRIZNO TECENJE - cross runing colors
for(te=0;te<=16;te++)
  {
  R=random(0,4)*63; // generate first blend
  G=random(0,4)*63;
  B=random(0,4)*63;
    
  R2=random(0,4)*63; // generate second blend
  G2=random(0,4)*63;
  B2=random(0,4)*63;
  
    for(n=0;n<NUM_LEDS;n++) // fire first color up and second down the string
      {
         leds[b] = CRGB(R,G,B);
         FastLED.show();
         leds[c] = CRGB(R2,G2,B2);
         FastLED.show();
	 delay(t*5);	
       
         b++; // increment 1st array coutner
         c--; // decrement 2nd array coutner
      }
      b=0; // array 1 begins at 0
      c=NUM_LEDS; // array 2 begins at the end
      delay(T);
  } 
}
```