wled/WLED #2974: Feature Request: Support saving multiple WiFi passwords

mickmister, 2022-12-26:
**Is your feature request related to a problem? Please describe.**

I'm always frustrated when I have to enter the WiFi password of every place I set up my LEDs, especially when I use them at the same place twice.

**Describe the solution you'd like**

Just like phones and PCs, I would like to store multiple WiFi credentials to allow the device to automatically connect to a router I connected to in the past.

**Describe alternatives you've considered**

I'm not sure of any alternatives. Certainly open to them :+1:

softhack007, 2022-12-26:
Similar to https://github.com/Aircoookie/WLED/issues/2845

blazoncek, 2023-02-05:
closed in favor of #2845

closed 2023-02-05 by blazoncek