wled/WLED #2976: Nightlight (sunrise) reboots esp32

jlambert121, 2022-12-26:
### What happened?

I was playing with the sunrise nightlight effect and notice that my ESP32 reboots about 4 minutes into a 5 minute sunrise every time.  This is running 0.14.0-b1 with a 60 LED WS2812B RGB strip, but I didn't ever try it on 0.13.3 so I can't say if it is a regression.

### To Reproduce Bug

Setup is 5 minute timer, target brightness 255, mode: sunrise.  60 LED strip, ESP32-dev.  Start the timer.

### Expected Behavior

Sunrise finishes, no reboot.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Let me know if there's information you'd like me to retrieve.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-27:
Can't duplicate.
Please get a debug build and monitor serial output using USB.

jlambert121, 2022-12-29:
I couldn't find anything in this repo or on the main website about the best way to do this.  I got no output using `esphomeflasher` or an ESP web tools console connection.  I couldn't find anywhere to enable additional logging or or update the log level either.  Could you point me to a good way to get this info?

devoh747, 2022-12-29:
How big is the power supply and are you routing the power thru the esp32
and then to the leds?  I ask because you are driving 60 leds and ramping up
to 255 for brightness..  if the power supply is too small it would cause
the voltage to drop and crash the esp32. I am speaking from experience
unfortunately.  Also if power is coming from the esp32 regulated power then
that too will not be enough and same problem.

Regards

  -Dave

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek