wled/WLED #2978: Led count and effects go wrong 》256

swampdogmash, 2022-12-27:
### What happened?

Tried to use strobe effect on a 5m x 60led/m ws2e12 strip and only 2/3 of the strip flashes. Reverted to solid effect and all leds lit up fine.

### To Reproduce Bug

Wemos d1 mini esp 8266 running wled 0.14.0 b1 with 300 leds configured. Large psu (5v 10a) with power injection and level shifter for data pin. Power restricted to 7000ma in wled.

Configure led options to be 300 less then single segment, 300 leds. Set strobe effect with single colour 1 as red and colour 2 as black. Only 2/3 approx of the leds light up (I cant count the first few hundred du to location but I can do the following:

Change the segment size to 256. The strobing leds reduce from 2/3 flashing to a few less. Gradually increase segment size from 256 one at a time.

257 adds an extra flashing led
258 adds an extra flashing led
259 no extra led
260 ..adds what seems to be led 259 flashing
261.. adds what seems to be led 260 flashing
262 no extra led
263 adds what seems to be led 261 flashing
264 adds what seems to be led 262 flashing
265 no extra led
266 adds what seems to be led 263 flashing
267 adds what seems to be led 264 flashing
268 no extra led
268 adds what seems to be led 265 flashing

It looks as though reducing the size also does the same thing.

There cant be 256 leds lit up (wish I could get to the start and end to count them!)  But it looks like:

256 x number of leds flash
255 no reduction in leds flashing
254 reduced by 1
253 reduced by 1
252 no reduction in leds flashing etc etc

Looks like 1 in 2 arent being addressed.

Sidenote:  I did this on a clean rather than upgraded wemos last night with totally different 2811 pixels and also the same seems to apply. Last night I just thought I was too drunk and a bit mental though.... lol. Last night I was using esp32 not 8266.

*****Simple test:  I've just tried configuring it as 9 leds in the led settings. With a single 9 led segment and only 6 leds are strobing. If I change to "solid" all 9 leds are controlled and light up appropriately. Scanner effect also only operates on 6 of the 9 configured leds as another example.

### Expected Behavior

The number of leds in the segment flash rather than 2/3 of them

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0.b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

swampdogmash, 2022-12-27:
In addition. Just tried configuring 450 leds in led settings and 450 in the segment and it is working as 300 used to work.

thewolf-oz, 2022-12-28:
I notice this when I specify the matrix to be Horizontal. When Vertical it is perfectly fine

swampdogmash, 2022-12-29:
I flashed a clean install and it didnt happen again. I have a feeling that this is to do with changing the number of LEDs upwards (on my fresh install of 0.14.0.b1) and with existing saved settings with the upgrade. Still cant quite figure out why though.

blazoncek, 2022-12-29:
Check your segments.

swampdogmash, 2022-12-29:
Hi, most def have checked segments. 1 segment, json looks to have the right number of leds. And text box also shows the right number of leds.

blazoncek, 2022-12-30:
Code wise there is no reason (apart from LED set-up or segment set-up) to do that.

swampdogmash, 2022-12-30:
Thats odd. I thiught I got it to work on a reflashed 0.14.0.b1 device but for example sinelon is only using 2/3 but sinelon rainbow is using the whole strip.

closed 2023-09-28 by blazoncek