wled/WLED #2979: Incorrect Light Capabilities set when LED strips is only On/Off

frenck, 2022-12-27:
### What happened?

When configuring the WLED strip with a single segment, only providing On/Off capabilities, WLED will determine its Light capabilities = 1 = RGB Light.

[image]

Dump from the `wled` Python library from the configured light:

Led info:
`
Leds(cct=False, count=1, fps=3, light_capabilities=<LightCapability.RGB_COLOR: 1>, max_power=850, max_segments=16, power=137, rgbw=False, wv=False, segment_light_capabilities=[<LightCapability.RGB_COLOR: 1>])
`

Segment state:
Segment(brightness=255, clones=-1, color_primary=(255, 160, 0), color_secondary=(0, 0, 0), color_tertiary=(0, 0, 0), effect=Effect(effect_id=0, name='Solid'), intensity=128, length=1, on=True, palette=Palette(name='Default', palette_id=0), reverse=False, segment_id=0, selected=True, speed=128, start=0, stop=1)
`

### To Reproduce Bug

Use the following configuration

[image]

### Expected Behavior

No to have RGB controls and communicate a Light Capability value of `0`.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-28:
@frenck can you confirm latest commit fixes the issue?

frenck, 2022-12-28:
I'm happy to confirm it on the next binary that is shipped? 🤷

ezcGman, 2023-01-04:
Just tested it with the latest master and the issue is resolved:
[image]
[image]

blazoncek, 2023-01-05:
If it is ok, please close.

frenck, 2023-01-05:
The HA release went out the door yesterday, meaning I have some time again. Will try to confirm today and close if resolved

PaoloTK, 2023-02-01:
> The HA release went out the door yesterday, meaning I have some time again. Will try to confirm today and close if resolved

Hi, any progress on this? Thank you.

blazoncek, 2023-04-24:
@frenck any updates? It has been a while...

blazoncek, 2023-05-27:
Closing as there is no feedback and the modification was made.

closed 2023-05-27 by blazoncek