wled/WLED #2980: Default OAT Pin

TysTies, 2022-12-27:
### What happened?

I checked the box in security and updates to Stop wireless updates. When 14.0 was released, I tried to update, and was ant able to get the box to uncheck. how do I uncheck it? this happened on multiple boards. 

### To Reproduce Bug

n/a

### Expected Behavior

n/a

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build2208222)

### Which microcontroller/board are you seeing the problem on?

ESP8266, ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

TysTies, 2023-01-02:
Hello?

Aircoookie, 2023-01-02:
In case you are referring to OTA lock, you'll need to enter the password set when enabling the lock in order to unlock it. The default value is "wledota". Please close if resolved.

closed 2023-01-03 by Aircoookie

TysTies, 2023-01-03:
Thank you. you might want to put that in yellow text underneath the box where you type the pin :)