wled/WLED #2983: WLED devices expose an impossible (or undocumented) light capability

frenck, 2022-12-28:
### What happened?

As per documentation, these are the light capabilities of WLED segments:

[image]

Source: https://kno.wled.ge/interfaces/json-api/#light-capabilities

However, that is untrue or a bug. I've got several ways to get `lc` values returned that exceed the documented bits, suggesting there is an additional bit in use that is currently marked as reserved/unused.

### To Reproduce Bug

Multiple reproductions:

- Configure a WLED device with a single segment "PWM CCT"; the `lc` value communicated is `14`, which exceeds the current documented set.

- Config a WLED device with a single segment "PWM RGB + CCT"; the `lc` value communicated is `15`, which exceeds the current documented set.

### Expected Behavior

To either not exceed the documented values or documentation that correctly provide information on the light capabilities.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Anything else?

I've currently assumed there is another bit involved and adjusted my library accordingly:

See: <https://github.com/frenck/python-wled/pull/876>

Ran into this while working on: <https://github.com/home-assistant/core/pull/75248>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ezcGman, 2022-12-28:
For some context: The line in question is this where the fourth/undocumented bit also gets set: https://github.com/Aircoookie/WLED/blob/main/wled00/FX_fcn.cpp#L754

blazoncek, 2022-12-28:
The docs say "reserved (expect any value)".
Use only the lower 3 bits.

Let's say (for now) that the upper nibble controls UI elements only.

frenck, 2022-12-28:
> Use only the lower 3 bits.

That turned out to be not valid. For example, an RGB+CCT with auto white, will provide the white bit, but since it is auto, that isn't applicable (which seems to be handled by the undocumented bit?) In the above case, the primary color is a tuple of 3 (R,G,B) and not 4 (R,G,B,W).

So it seems to have significance, and the WLED UI seems to care about it as well.

> Let's say (for now) that the upper nibble controls UI elements only.

It also changes the data send over the line.

blazoncek, 2022-12-28:
The relevant code:
```c++
bool whiteSlider = (aWM == RGBW_MODE_DUAL || aWM == RGBW_MODE_MANUAL_ONLY); // white slider allowed
if (bus->isRgbw() && (whiteSlider || !(capabilities & 0x01))) capabilities |= 0x08; // allow white channel adjustments (AWM allows or is not RGB)
```

blazoncek, 2023-02-16:
Undocumented capability removed in #3073

closed 2023-02-16 by blazoncek