wled/WLED #2984: "White" channel only segments confusingly provide 2 or 3 sliders

frenck, 2022-12-28:
**Is your feature request related to a problem? Please describe.**

Currently, when configuring a device with a single segment with a single white channel, creates a confusing UI to control and a confusing API to work with.

In the end, the case described above is basically a dimmable LED strip (single color, assumingly just "white").

The point is, WLED will give you 3 sliders for controlling this LED strip, providing the same functionality, but are fully decoupled:

- The master slider for the brightness.
- The brightness slider for the segment.
- The white value slider for the segment.

All of these do the same thing in this case: Controlling the brightness; which makes no sense and is (in my opinion) highly confusing.

**Describe the solution you'd like**

The master brightness makes sense in a way, as in, that is part of how WLED works in general.

I would suggest removing the slider for the white values in the case as described above and only providing a brightness slider for the segment (omitting the white value).

**Describe alternatives you've considered**

I've not, really. For external control, I've considered sending the same values for brightness and white value (so external control is at least logical).

**Additional context**

[image]

Ran into while working on: <https://github.com/home-assistant/core/pull/75248>

Thank you for your ideas for making WLED better!

Aircoookie, 2023-01-02:
I agree that having 3 sliders for apparently the same purpose in typical use cases (1 segment, solid effect) is confusing.

However, keep in mind that even "White only" segments can take advantage of segment color slots (consider the Blink effect blinking between maximum brightness and a dimmer background color value)

ezcGman, 2023-03-09:
So this is solved now? :)

Trying to get everything closed and solved to finally get this added to HA :)
https://github.com/home-assistant/core/pull/75248

To finally have CCT support for WLED in HA :)

Thanks!

frenck, 2023-03-09:
This is not blocking the referenced PR in Home Assistant Core.

blazoncek, 2023-03-09:
The core of this issue is not "fixed" and probably will never be.
But.
I have added UI logic to hide one of the sliders (segment opacity/brightness) if only one segment is used. So the number of possible options to change brightness is reduced from 3 to 2 sliders (one being global brightness the other being "color").

ezcGman, 2023-03-09:
Yeah, I'm with you. I like the improvement for this now! Thx!

closed 2023-06-07 by blazoncek