wled/WLED #2989: ESP32 White channel missing

martin072, 2022-12-29:
### What happened?

When compiled for a ESP32 (Lolin s2 mini) there is no "white channel slider" on the UI, whilst compiled on the eps8266 (d1_mini), it shows correctly. 

Furthermore, the HEX input (when enabled) shows only RGB value and not RGBW (when selected strip is an RGBW)

### To Reproduce Bug

See above.

### Expected Behavior

White channel should be visible when configured as RGBW strip on ESP32

### Install Method

Self-Compiled

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

martin072, 2022-12-29:
**ESP32**

[image]

**ESP8266**

[image]

softhack007, 2022-12-29:
I think you need to set "auto-calculate white" in LED preferences
[image]

After adjusting the settings, I have performed a reboot (info panel), and the "white balance" slider appeared.

Please check your LED preferences.

blazoncek, 2022-12-29:
FYI: S2 is not *regular* ESP32.

martin072, 2022-12-29:
Hmm. Seems to be intermitted. When doing a reboot, the slider is back..

blazoncek, 2022-12-30:
Then please close.
The appearance of white slider depends on how white handling is selected and has been tested numerous times.

closed 2022-12-30 by martin072

blazoncek, 2023-01-06:
Read the docs and you'll notice **a lot** of differences between those chips.
For start, it is a single core MCU.

softhack007, 2023-01-06:
> > FYI: S2 is not regular ESP32.

> Can you elaborate on this? 

ESP32-S2 is not a regular "classic ESP32".
* It has a different Processor, so you need a firmware that was build for -S2
* It only has one core (instead of two)
* It is slower (240Mhz vs. 160Mhz)
* It Only supports 4 LED outputs, instead of two
* It is too slow for audioreactive usermod (ok, on the edge actually)

and above all that: WLED support for -S2 (and for -S3 and -C3) is *experimental* and still *bleeding edge*. It may work or not work. So better if you re-think your idea of exchanging all your D1 Mini's to Lolin's ESP-S2...

blazoncek, 2023-01-06:
A side note: C3 works much better as a substitute for ESP8266.

martin072, 2023-01-06:
Thanks for all the info! 🙏