wled/WLED #2992: 0.14.0-b1 DoS / crash on excessive http requests

sh00bx, 2022-12-30:
### What happened?

I noticed I can reliably crash WLED 0.14.0-b1 on ESP32 by sending a larger amount of http requests to the web GUI.
The device becomes unavailable after less than 100 requests sent somewhat high frequency (10 concurrent, 10s timeout)

### To Reproduce Bug

Flood http, e.g. using gobuster:
`gobuster dir -u http://wled -w /usr/share/seclists/Discovery/Web-Content/raft-medium-words.txt`

### Expected Behavior

WLED should stay online and a) keep responding to http requests or ideally b) blacklist the source to prevent DoS

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
# gobuster dir -u http://wled -w /usr/share/seclists/Discovery/Web-Content/raft-medium-words.txt
===============================================================
Gobuster v3.4
by OJ Reeves (@TheColonial) & Christian Mehlmauer (@firefart)
===============================================================
[+] Url:                     http://wled
[+] Method:                  GET
[+] Threads:                 10
[+] Wordlist:                /usr/share/seclists/Discovery/Web-Content/raft-medium-words.txt
[+] Negative Status codes:   404
[+] User Agent:              gobuster/3.4
[+] Timeout:                 10s
===============================================================
2022/12/30 17:43:09 Starting gobuster in directory enumeration mode
===============================================================
Progress: 57 / 63088 (0.09%)[ERROR] 2022/12/30 17:43:19 [!] Get "http://wled/login": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:19 [!] Get "http://wled/search": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:19 [!] Get "http://wled/cache": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
Progress: 60 / 63088 (0.10%)[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/include": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/category": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/logout": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/comment": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/report": context deadline exceeded (Client.Timeout exceeded while awaiting headers)
[ERROR] 2022/12/30 17:43:20 [!] Get "http://wled/tag": context deadline exceeded (Client.Timeout exceeded while awaiting headers)

# ping wled
PING wled (x.x.x.x) 56(84) bytes of data.
From y.y.y.y icmp_seq=1 Destination Host Unreachable
From y.y.y.y icmp_seq=2 Destination Host Unreachable
From y.y.y.y icmp_seq=3 Destination Host Unreachable
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2022-12-30:
ESP is a MCU with limited resources and yes, hitting it with multiple requests will bring it down.
There is nothing that can be done.
Adding IP addresses to a blacklist will deplete RAM even further.

sh00bx, 2022-12-30:
Thanks, you're right, at first I was surprised that my tasmota ESP8266 held up against flooding http much longer than the WLED ESP32. But turns out that after a while (~5000 requests) the tasmota went down as well and even had its config wiped out 😨 
So I guess I'm happy with WLED "just" crashing.

closed 2022-12-30 by sh00bx