wled/WLED #2995: Connecting ESP32-S2 Saola to wifi causes bootloop

the1glorfindel, 2023-01-02:
### What happened?

After building and uploading latest release (0.14.0-b1) esp32-S2-Saola goes into bootloop after connecting it to my wifi network. It seems to work indefinitely if I only control it through the AP.
One other factor seems to be the specific lights I'm using, since the issue was not present until I connected them. I controlled other random strings that I had around, but only after I got my new ones in ([https://amazon.com/dp/B097BX7P3V](https://amazon.com/dp/B097BX7P3V)) did it start bootlooping.

### To Reproduce Bug

1. Build latest release in PIO
2. upload to ESP32-S2_SAOLA
3. Wire up lights (possibly the specific ones I had?)
4. Enter settings via AP and configure for home network
Results for me was lights go out, and serial monitor indicates bootloop.

### Expected Behavior

Normal operation

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32-S2

### Relevant log/trace output

```shell
--- Terminal on COM4 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x108
load:0x4004c000,len:0xa08
load:0x40050000,len:0x26fc
entry 0x4004c174
Ada
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146c04: f01d1662 c0004136 22820020
Core  0 register dump:
PC      : 0x40146c0a  PS      : 0x00060031  A0      : 0x40028c0d  A1      : 0x3ffc3cf0
A2      : 0x3ffc30f4  A3      : 0x3f416000  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000020  A8      : 0x80025af0  A9      : 0x00000020
A10     : 0x3ffd9948  A11     : 0x3ffd9948  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x3ffc3100  A15     : 0x3ffd9b40  SAR     : 0x00000014  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0xffffffff  LEND    : 0x00000000  LCOUNT  : 0x40028aed  
Core  0 was running in ISR context:
EPC1    : 0x400e018f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40146c07:0x3ffc3cf00x40028c0a:0x3ffc3d30 0x4003011e:0x3ffe45e0 0x4002fde0:0x3ffe4600 0x40031cba:0x3ffe4620 0x40026d86:0x3ffe46a0 0x400db7b1:0x3ffe46e0 0x40100a29:0x3ffe4710 0x40100d99:0x3ffe4730 0x4010119e:0x3ffe4770 0x40101783:0x3ffe4810 0x40103a69:0x3ffe4870 0x401003c0:0x3ffe48c0 0x400e15c5:0x3ffe49f0 0x401407cd:0x3ffe4a10 0x4013adfa:0x3ffe4a30 0x400c91d8:0x3ffe4ab0 0x4014600f:0x3ffe4af0 0x400c6b93:0x3ffe4b10 0x400a760e:0x3ffe4b30 0x400c1667:0x3ffe4b80 0x400cecdc:0x3ffe4bd0 0x400cd0d5:0x3ffe4c20 0x400cd1a5:0x3ffe4c70 0x400cd365:0x3ffe4cc0 0x400c5fc1:0x3ffe4ce0 0x400c6055:0x3ffe4d10 0x400c6682:0x3ffe4d30

ELF file SHA256: 0000000000000000

Rebooting...
ESP-ROM:esp32s2-rc4-20191025
Build:Oct 25 2019
rst:0x3 (RTC_SW_SYS_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40027325
SPIWP:0xee
mode:DIO, clock div:1
load:0x3ffe6100,len:0x108
load:0x4004c000,len:0xa08
load:0x40050000,len:0x26fc
entry 0x4004c174
Ada
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x40146c04: f01d1662 c0004136 22820020
Core  0 register dump:
PC      : 0x40146c0a  PS      : 0x00060031  A0      : 0x40028c0d  A1      : 0x3ffc3cf0  
A2      : 0x3ffc30f4  A3      : 0x3f416000  A4      : 0x00000001  A5      : 0x00000000  
A6      : 0x00000001  A7      : 0x00000020  A8      : 0x80025af0  A9      : 0x00000020
A10     : 0x3ffda25c  A11     : 0x3ffda25c  A12     : 0xffffffff  A13     : 0x00000000  
A14     : 0x3ffc3100  A15     : 0x3ffda09c  SAR     : 0x00000014  EXCCAUSE: 0x00000000
EXCVADDR: 0x00000000  LBEG    : 0xffffffff  LEND    : 0x00000000  LCOUNT  : 0x40028aed  
Core  0 was running in ISR context:
EPC1    : 0x400e018f  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x00000000

Backtrace:0x40146c07:0x3ffc3cf00x40028c0a:0x3ffc3d30 0x4003004e:0x3ffe0980 0x4002b4ae:0x3ffe09a0 0x40026d6a:0x3ffe09c0 0x400db7b1:0x3ffe0a00 0x400dcd22:0x3ffe0a30 0x40146f21:0x3ffe0a50 0x400dd701:0x3ffe0a70 0x400dbf37:0x3ffe0ae0 0x400dc410:0x3ffe0b30 0x400dcbad:0x3ffe0ba0 0x400dbc7a:0x3ffe0bc0 0x4011fb61:0x3ffe0bf0 0x4011fdf7:0x3ffe0c30 0x4011de74:0x3ffe0c50 0x4011c7a6:0x3ffe0c70 0x4014aa09:0x3ffe0c90 

ELF file SHA256: 0000000000000000

Rebooting...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

the1glorfindel, 2023-01-02:
Correction to the above: it seems the issue also occurs with no leds connected. However, the issue appears when I set the led string to 300 leds in length

blazoncek, 2023-01-02:
S2 is not yet officially supported.

You will need to tweak settings, use different core, introduce new libraries to make it work. If you figure a consistent way for S2 to work flawlessly make a PR with your findings.

wpbirney, 2023-01-04:
Curious why -S2, -S3 or -C3 wouldn't be supported? They all have everything needed and full support of all esp32 based mcu's in esp-idf. I'm not familiar with the WLED codebase but i do not see any reason (from an esp-idf standpoint) why it wouldn't be transparently supported. Even with the ardunio libraries i assume you are using, which are all built on top of esp-idf.

softhack007, 2023-01-04:
> Curious why -S2, -S3 or -C3 wouldn't be supported?

Your view might be too simple. There is a huge number of differences between "classic ESP32" and the new chips. 

Like different bootloaders needed depending on exact PSRAM and FLASH interfaces, different architectures and in-chip units(number of RMT units is just one example), different GPIO numbers and capabilities. 

It took us (2-3 spare-time open source developers) around three month to get the basics of WLED working on a few devices we had at hand. 

On top of that, we had to fight with a huge load of unexpected crazy bugs in the newer ESP-IDF v4.4.x. 

And we still cannot say "it's running reliability". That's why s why -S2, -S3 or -C3 wouldn't be supported formally in the latest release. We still considers support for these new MCU's experimental and bleeding edge.

wpbirney, 2023-01-04:
I have and still do work with esp-idf a lot, my view is anything but simple. That said i can not reproduce the bug of 300 or more LED's on the S2's or S3's i have on hand. The differences on the -s2 -c3 -s3 are pretty transparent as far as esp-idf is concerned. The bug he listed is pretty commonly seen for flash corruption and might not be software related.

softhack007, 2023-01-04:
You are welcome to help. Make a PR if you know how to fix the problem.

blazoncek, 2023-01-05:
The major issue (with my test unit) was LittleFS access and AsyncWebServer.
Both not stable at all.

Jason2866, 2023-01-25:
LittleFS is rock solid. Can say this since we use it now for over a year (or already 2 Years) as default FS on every ESP32 (including S2, S3 and C3) for Tasmota. The issue is probably caused from AsyncWebServer. We do not use AsyncWebServer in Tasmota.

jayzakk, 2023-02-20:
I actually had the same problem, it usually happens in `wled.cpp` at `WiFi.config()` after the `WiFi.disconnect()` (line#621).
The workaround for that: you have to add a `delay()` (I used 1000 as usual) after the `disconnect()` and before any other WiFi operation, so things (threads and irqs) can settle/clean up. That's some very old bug and looks it's back again under certain cicumstances.

In general, this happened on **all** of my S2's as soon as enabling NTP; and when trying to run with `WLED_DEBUG` defined (bummer!).

I didn't see this problem on C2 and S3.

As I think the delay(1000) isn't anything cool to put into an PR - but maybe it's the only possibility - I didn't create one ^^

Going to look after some more of spurious reboots after vacation and when the S2-devkit arrives - it's absolutely impossible to debug anything with the S2-mini. LOLIN HW devs should get a kicked-in-their-a** for neither exposing GPIO41+42 to be able to use JTAG nor native GPIO43+44 of UART0 (but 37+39 instead).

Regards.

blazoncek, 2023-04-24:
I can now confirm that C3 and S2 are running issue free for over a month after changing PIO ESP32 core to 5.2.0 and using ARDUINO_CDC_ON_BOOT=0 which seems mandatory for successful operation on S2 and C3.

closed 2023-04-24 by blazoncek

johan149, 2023-08-24:
> I can now confirm that C3 and S2 are running issue free for over a month after changing PIO ESP32 core to 5.2.0 and using ARDUINO_USB_CDC_ON_BOOT=0 which seems mandatory for successful operation on S2 and C3.

Can you share the PIO environment settings for S2 compilation... I'm also facing some issues when connecting to my WIFI.