wled/WLED #2997: Are there any plans to support keyboards?

TriplEight, 2023-01-04:
**Is your feature request related to a problem? Please describe.**
I'm always frustrated when a RGB lighten keyboard doesn't have the convenient way to control these lights and behaves as it will :rainbow: .

**Describe the solution you'd like**
It would be great to have something like https://github.com/libratbag/libratbag or https://gitlab.com/CalcProgrammer1/OpenRGB but more user-friendly. And, mainly, be compatible with the entire fleet of LED stuff you already support!

**Describe alternatives you've considered**
https://github.com/libratbag/libratbag and https://gitlab.com/CalcProgrammer1/OpenRGB. The latter has a lot of supported devices, including mice, LED-lighten wotherboards and RAM chips. But AFAIU nothing of the stuff you support.
That would be great to marry these swarms of devices with your neat UI.

CalcProgrammer1, 2023-01-05:
OpenRGB can output E1.31 which WLED can receive, so you can control WLED as well as your keyboard with OpenRGB and OpenRGB's Effects Plugin.

If WLED offers a way to stream effects from the WLED effects engine out to other devices I'd be up for making a plugin for OpenRGB to take in that stream and feed it to other devices.  We already have an E1.31 receiver plugin so you can control OpenRGB devices with E1.31 (which means you can also use OpenRGB as an E1.31 repeater if you'd like to control both PC peripherals and WLED devices from one E1.31 source).

blazoncek, 2023-01-05:
WLED can output DDP.

CalcProgrammer1, 2023-01-06:
Never heard of DDP, but looking at the protocol documentation here
http://www.3waylabs.com/ddp/ I think it's something that we should probably
implement in OpenRGB (for sure output, possibly also input as a plugin).
The bandwidth advantages over E1.31 might be worth upgrading my fleet of
ESPixelStick controllers to WLED as my Christmas lights lag sometimes.

deirisu, 2023-12-10:
I just discovered this discussion so I am sorry if i bring it back to life lol 
I don't know if you know about Signal RGB which has a native wled integration.