wled/WLED #2998: Usermod "Battery" features

itCarl, 2023-01-04:
First of all this isn't a real Bug neither a FR. sorry. I just want to get in touch with people that want to improve the Usermod.

I have seen that recently many good improvements have been added to the usermod. In fact, I am aware of the "bugs"/missing features.
That's why I started to write a software to simulate a battery (later battery packs will be supported).

The Software is still far from finished but a start has been made. The main goal, however, is to use the software to better understand and analyze batteries in order to significantly improve the usermod.
I think the software becomes very interesting especially when you have varying currents and you want to look at the runtime when a certain wled effect is activated. In my head I also have such features as "SOH, State Of Health", "Battery types" or "Smart battery calibration". Knowing these parameters from a battery for just a usermod might be a bit too much though.

Some Screenshots
---
[image]
[image]

This Software is still kinda basic. But can fully calculate parameters for static loads.

softhack007, 2023-01-04:
Hi @itCarl, I just added some initial support for 
LiPo batteries, as I have a device with built-in LiPo connector/charger.

I found two interesting things:
* as ESP32 cannot measure above ~2.9V, these devices all use a small "voltage devider" circuit (basically two resistors of the same value). So measured voltage must be multiplied by 2.
* LiPo cells seems to have a different  voltage / capacity curve that is non-linear at its "edges",  see https://blog.ampow.com/amp/lipo-voltage-chart/

I'm happy with my results now, and I think it will also work on other esp32 with build-in LiPo connector.

Actually I don't really understand how the voltage measurements on ESP8266 work, as the original calculation seems to use some "hidden magic numbers", and the results on esp32 looked very wrong...

softhack007, 2023-01-04:
In general I think it's not correct to assume that voltage and capacity have a linear relation. Might be different for 18650, however seems to be different for other types of battery, especially the LiPo cells.

softhack007, 2023-01-04:
Another thought on the "remaining battery time". 
* the downside of using `strip.currentMilliamps` will be that  this value is momentary. It can change from one millisecond to the other - literally. And it assumes that there is a linear relation between LED power usage and led brightness, which generally is not true.

* another option could be
* - monitor how battery capacity (batterylevel) drops during 10 minutes or so.
* - calculate how long it would take (ar the same rate) to go down to 10%.
* - might be more reliable, and you can even do that without knowing the total ampere that the battery can store.
* - would even work with worn-out batteries that do not reach their design capacity any more

itCarl, 2023-01-05:
> In general I think it's not correct to assume that voltage and capacity have a linear relation. Might be the case for 18650 (I don't know), however seems to be different for other types of battery, especially the LiPo cells.

That fact that the Charge/Discharge Curve of Batteries isn't linear is 100% correct. At the time i created this usermod i didn't know any better i just needed a rough estimate of the battery charge / voltage.

> Another thought on the "remaining battery time".
> 
> * the downside of using `strip.currentMilliamps` will be that  this value is momentary. It changes from one millisecond to the other - literally. And it assumes that there is a linear relation between LED power usage and led brightness, which generally is not true.
> * another option could be
> * * monitor how battery capacity (batterylevel) drops during 15 minutes or so. Like store a battery level value every 5 minutes in a small array with 4 entries, and always use the oldest value.
> * * calculate how long it would take at the same rate (that you get for last 15+x min) to go down to 10% -> remaining battery time.
> * * might be more reliable, and you can even do that without knowing the total ampere that the battery can store.
> * * would even work with worn-out batteries that do not reach their design capacity any more

A really huge problem ist that the battery voltage drops when a load is applied. As you can imagine that is a huge Problem because the voltage is the base parameter that all calculations are based of rn.
For a reliable measurement, the LEDs must be switched off via a transistor so that the battery voltage can settle. Only then can the voltage be measured.
If you want to achive a near perfect estimation of the battery capacity you would have to build a piece of Hardware that does exactly that. For example a [SparkFun Coulomb Counter Breakout - LTC4150](https://www.sparkfun.com/products/12052) could do that. That method of estimating the battery capacity is called "[Coulomb counting](https://en.wikipedia.org/wiki/State_of_charge#Current_integration_method)". It's based on current / how many C's got discharged. The downside is that it has to be re-calibrated on a regular basis.

softhack007, 2023-01-05:
> A really huge problem ist that the battery voltage drops when a load is applied. As you can imagine that is a huge Problem because the voltage is the base parameter that all calculations are based of rn.

🤔 hmmm good point. Will this drop be there while the load _stays_ connected, or is it just happening _initially_ but then it settles and comes back to "normal voltage"? 
Looking from signal processing perspective - If its an initial short-time voltage drop, then maybe a Kalman filter can compensate the drop, and the Kalman "quality" parameter could help to ignore measurements with "low confidence".

itCarl, 2023-01-05:
>   #ifdef USERMOD_BATTERY_USE_LIPO
    // LiPo "1S" Batteries should not be dischared below 3V !!
    #define USERMOD_BATTERY_MIN_VOLTAGE 3.2f
  #else
    #define USERMOD_BATTERY_MIN_VOLTAGE 2.6f
  #endif

@softhack007 - I saw your change in the last commit [35832b0](https://github.com/Aircoookie/WLED/commit/35832b07b9faba9d2cd13431e16d6b874975c875) and i'm not sure that it is 100% correct. 1.
As example, a TP4056 charging module that also features a discharge protection circuit(chip (the [DW01](https://datasheet.lcsc.com/szlcsc/1901091236_PUOLOP-DW01A_C351410.pdf)) have a "Overdischarge Protection Voltage" at 2.6 Volts. Thats why i choose the 2.7 v also the default for minBatteryVoltage
The Secondthing is that the Datasheet of the [18650 Batteries that i](https://www.akkuteile.de/lithium-ionen-akkus/18650/molicel/molicel-inr18650-m35a-3500mah-10a-lithium-ionen-akku-3-6v-3-7v_100833_2843) use have a Discharge voltage of 2.5 Volts

softhack007, 2023-01-05:
@itCarl maybe it really depends on the battery type. I was thinking of these "Chinese flat rechargeable" batteries like the one from makerfocus, that are very popular for RC drones and small DIY devices. These usually have a warning note like "don't discharge below 3V". 
Even when 18650 might also be LiPo-based, it can be that 18650 LiPo have different characteristics.  

Most "flat rechargeables" have a builtin under-voltage protection circuit (notice the small chips under yellow tape), so it might be that the T7-v15 does not need additional protection.

I can double-check later what is correct, but just to be save, I think that 3.2V as default low voltage limit might be better. Any way users can change that as they want, its just a preset.

softhack007, 2023-01-05:
This is another typical "flat LiPo" from China:

[image]

[image]

itCarl, 2023-01-05:
> Also The battery Discharge CutOff is at 3V but the protection circuit has a discharge detection at 2.5v. I'm confused big times 😕.

Do you have a multimeter or other device to measure the battery voltage?
It would be interesting to see what voltage the battery actually discharges to.

softhack007, 2023-01-05:
> Do you have a multimeter or other device to measure the battery voltage?

Yes, both 😉 I can connect a battery and let it run out, then measure remaining voltage - I'll tell you tomorrow or on the weekend.

itCarl, 2023-01-06:
@softhack007 Very nice, Thanks!

I just thought about the calibration feature/value. Does it make sense to calculate it instead of having a correction value entered manually? So what i mean by that correctionValue = measured voltage by multimeter or something / sensed voltage by the microcontroller. And then the correctionValue gets multiplied by the calculated voltage.

I'm also thinking about to add support for a transistor switch pin. It would be nice to lower the leakage current from the voltage divider circuit. But thats probably to 'advanced' for a lot users.

softhack007, 2023-01-10:
> Do you have a multimeter or other device to measure the battery voltage?
> It would be interesting to see what voltage the battery actually discharges to.

Hi, I just manged to "watch a LiPo die", and took Multi Meter measurements during the last minutes.
* At 3.0V, the ESP32 start to become unstable, and I get random reboots (brownout or watchdog reset)
* The on-board (ESP32 T7) battery circuit shuts down at 2.9V, so I could not power it on again with the battery below 2.9V
* The battery itself switches off at 2.6V
* Time between 2.9V and 2.6V is less than 4 minutes, voltage goes down very quickly

itCarl, 2023-01-10:
I really appreciate the work you have done. It has answered some questions about batteries for me. And I think this will be a good source of information for the future.

> Edit: a simple Kalman implementation is here: https://github.com/denyssene/SimpleKalmanFilter

That sounds and looks like a good idea 👍 I've seen a similar smoothing method. It looked very very simple and easy to use without any libs.
But I haven't had time to take a closer look yet.

softhack007, 2023-01-10:
> It looked very very simple and easy to use without any libs.

A  very simple (but effective) smoothing method is using an exponential filter, like this:
```C++
constexpr float alpha = 0.1; // between 0 and 1, to control strength of the filter
static float filteredValue = 0;   // smoothed output - kept between filter runs

float raw = analogRead(34); // raw value
filteredValue = filteredValue + alpha * (raw-filteredValue);  // apply exponential filtering
```

We use this simple method a lot in the audioreactive usermod.

itCarl, 2023-01-10:
Interesting. The code snippet i found looks kinda similar and is propably doing the identical think:

`x = (99.0f * x + value) / 100.0f`

softhack007, 2023-01-10:
> `x = (99.0f * x + value) / 100.0f`

yes its basically doing the same, just that my "audioreactive filter" variant is a bit more efficient (avoids division), and it keeps more precision (avoids the *99 and /100). There are many ways to write an exponential filter.

itCarl, 2023-01-18:
Okay. So I will implement your method of smoothing if it's ok for you.

I'm struggling come up with a charging indicator.

softhack007, 2023-01-21:
> Okay. So I will implement your method of smoothing if it's ok for you.

Hi, sure go ahead, that's fine with me. The tricky part might be to find a good smoothing parameter. Due to low sampling rate, I'd suggest to start with 0.25 or 0.1, and then try reducing the value until you are happy with the smoothed results.

Very busy right now, so It'll take some time before I can review you latest code...

Cheers ✌️

Easyrunner, 2023-06-30:
Hey guys, I'm running the battery usermod and select this auto off function. 
What happen when this switches to off? How it will switch on again? 
I have now the problem, that the threshold was set to 3.0V, but battery was not fully empty, so the Wemos was always powered. Now I charge the battery, but the Wemos seems "dead"
Does it needs a power cycle then by Reset or short power-less?

Thx!

itCarl, 2023-07-01:
Hi @Easyrunner,
Thank you for using this usermod.
It does not switch on automatically after you charged the battery.
The battery has to be charged above the autoOffThreshold in order to turn it manually on again.

JohnShen04, 2024-07-17:
你好，我是一个新手，请问是否可以请教一下IIC通信的代码呀？调试时貌似通信地址不正确，希望能得到你们的帮助，十分感谢！

softhack007, 2024-07-17:
> 你好，我是一个新手，请问是否可以请教一下IIC通信的代码呀？调试时貌似通信地址不正确，希望能得到你们的帮助，十分感谢！

Could you please translate to englisch?

JohnShen04, 2024-07-17:
> > 你好，我是一个新手，请问是否可以请教一下IIC通信的代码呀？调试时貌似通信地址不正确，希望能得到你们的帮助，十分感谢！
> 
> 你能翻译成英语吗？

Hello, I am a beginner. May I ask for the code for IIC communication? It seems that the communication address is incorrect during debugging. I hope to receive your help. Thank you very much!

itCarl, 2024-07-17:
I think we should close this issue after we have helped JohnShen04. I don't want this issue/discussion to turn into a Battery usermod help ticket. People should create separate issues for their questions. Otherwise this Ticket/Issue will get too confusing.

JohnShen04, 2024-07-17:
> 我认为我们应该在帮助 JohnShen04 之后关闭这个问题。我不希望这个问题/讨论变成电池 usermod 帮助票。人们应该为他们的问题创建单独的问题。否则，这张票/问题会变得太混乱了.

Yes .I think you're right. I will delete my question, but before that, can I ask what the IIC address is? I think my problem lies here/(ㄒoㄒ)/~~

JohnShen04, 2024-07-17:
> > 我认为我们应该在帮助 JohnShen04 之后关闭这个问题。我不希望这个问题/讨论变成电池 usermod 帮助票。人们应该为他们的问题创建单独的问题。否则，这张票/问题会变得太混乱了.
> 
> 是的。我认为你是对的。我将删除我的问题，但在此之前，我可以问一下 IIC 地址是什么吗？我觉得我的问题就在这里/（ㄒoㄒ）/~~

I will appreciate it if  you can give me a helping hand  : )

itCarl, 2024-07-17:
@JohnShen04 i dont quite know what u mean by LLC (l2c?) Address.

> Hello, I am a beginner. May I ask for the code for IIC communication? It seems that the communication address is incorrect during debugging. I hope to receive your help. Thank you very much!

JohnShen04, 2024-07-18:
> > > 我认为我们应该在帮助 JohnShen04 之后关闭这个问题。我不希望这个问题/讨论变成电池 usermod 帮助票。人们应该为他们的问题创建单独的问题。否则，这张票/问题会变得太混乱了.
> > 
> > 
> > Yes .I think you're right. I will delete my question, but before that, can I ask what the IIC address is? I think my problem lies here/(ㄒoㄒ)/~~
> 
> 我不确定我是否完全理解这个问题。一些想法
> 
> * 如果使用 esp32，任何 GPIO 引脚都可以做 I2C。I2C 引脚设置位于 WLED 的“usermod 设置”页面中。
> * 如果您使用的是 8266，通常引脚 4 和引脚 5 是 I2C。
> * I2C引脚与电池usermod无关;Battery UserMod 与模拟读数一起工作。
> * 也许你说的是另一个 usermod，比如 https://github.com/Aircoookie/WLED/tree/0_15/usermods/MAX17048_v2 ？

OK.I see.Thank you very much!

closed 2024-07-18 by itCarl