wled/WLED #3007: WLED ESP32 - Unrecoverable Boot Issue

rac146, 2023-01-06:
### What happened?

It seems somehow one of my WROVER-E ESP32 boards runnings WLED started to experience the behavior outlined in this detailed issue:
https://github.com/espressif/esp-idf/issues/7994

No matter what, it's stuck in a boot loop - no way of erasing, write_flash_status, etc. - any attempt results in no change to the board.

Not sure how this happened - the boxes are powered by a switch so it could have been something related to power on/off. My lights are also running off GPIO12, which apparently can cause some adverse side effects as well. (going to switch to GPIO14). The only thing I remember was that after WLED working for some time, the ESP32 would stop saving any changes to the WLED settings (it would always revert back to a previous state). Also a re-install would result in no change (previous version would still be on the chip).

Towards the bottom of the issue there is a recommendation for a bootloader update - is there anything WLED can do (or has already been done) to implement this?

### To Reproduce Bug

Can't reproduce

### Expected Behavior

ESP32 should not go into this state

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.13.3 (build 2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-06:
If you can't erase flash or upload some other firmware then your ESP is toast. Dead. Kaput.
Just get a new one.

rac146, 2023-01-06:
Agreed - will replace.

It is unfortunate that the ESP32 WROVER-E's can brick themselves - but if there's a way to implement some of the changes suggested by Espressif into WLED to prevent future failures, I think that should be looked into..

blazoncek, 2023-01-06:
It is most likely a bad flash, nothing more. I have a few of those myself.

rac146, 2023-01-06:
But there's an underlying issue to this. In my case, it wasn't a bad flash - I was noticing WLED was not persisting any settings. After thorough research, I can confirm this is what I experienced to my ESP32 WROVER-E. Both SRP0 and SRP1 bits got switched to 1 on a power off, essentially bricking the device.

Through Espressif's recommendations (from the above Github issue), this is an issue with their chips. There is guidance on how to prevent this and this issue is merely opened to discuss if any of their recommendations can (or should) be incorporated into WLED. I would like all owners of WROVER-E's to be able to run WLED without this potential problem!

blazoncek, 2023-01-06:
Try DOUT flash mode if you haven't. I recovered some units with that.
All in all it is just 5EUR for a single ESP. What's the fuss?

You can try to use newer ESP core by compiling WLED yourself. Update platformio.ini and make sure you use most recent bootloader.
Be warned: I've had bad experience with latest ESP core and libraries used with WLED.

rac146, 2023-01-06:
WIll give it a try!

I do have a lot of WROVER-E's running WLED so my concern is others are going to be bricking in the future..

I am curious on Espressif's recommendation:
_"For new products, we recommend IDF version v4.0.4 (for release/v4.0), v4.3.2 (for release/v4.3), v4.4 and later, to mitigate this issue. "_

Not very familiar with the bootloader - does WLED use these IDF versions in its bootloader? If so, what version is currently being used?

closed 2024-04-23 by github-actions[bot]