wled/WLED #3008: esp8266 sound reactive not found

DJprasenjit, 2023-01-07:
sound reactive effects not work on esp8266. there is no option on settings. only hw pins shows on usermod but no use of that.
please make it work on esp8266 device

softhack007, 2023-01-07:
audioreactive usermod is not possible on 8266, it only works on ESP32. 8266 is simply too slow for audio processing (FFT).

Actually a minimal usermod that only supports UDP sound sync - no microphone - is in work.

More info is here:
* https://github.com/atuline/WLED/issues/230 
* https://github.com/blazoncek/WLED/issues/37

softhack007, 2023-09-06:
There will be basic sound sync support for 8266 in 0.14.1.
Full audioreactive functionality is not possible on 8266.

closed 2023-09-06 by softhack007