wled/WLED #3011: Can not add WLED to Home Assistant  0.14.0-b1

geeneo, 2023-01-09:
### What happened?

When I add to Home Assistant I get the  following error  "This WLED device uses CCT channels, which is not supported by this integration"

### To Reproduce Bug

[image]

### Expected Behavior

I don NOT have white balance enabled

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

geeneo, 2023-01-10:
> That is an issue of HomeAssitant, not WLED. The developers choose to disallow such configuration.

Thx blazoncek.

Do you know if they will fix it and do you know if there will be a workaround ?

Is this only me having this "problem" ?

blazoncek, 2023-01-10:
How should I know? Ask or open an issue on Home Assistant Github.
They seem to not like the implementation of CCT WLED choose.
And no, you are not alone in this.

As a workaround you may choose different calculations of CCT in WLED if any will help present WLED as a non-CCT device to HA.

geeneo, 2023-01-10:
Hi blazoncek

Can you please give me more info on the workaround, what and where do I need to change ? 

Thx for the help!

blazoncek, 2023-01-10:
https://kno.wled.ge/features/cct/

3ative, 2023-02-04:
I had this too.
What fixed it was a WLED reset

mrosseel, 2023-02-27:
seems like HA cannot determine whether it's RGB+CCT or CCT which is why they disabled all CCT handling to prevent unexpected behaviour. See PR below where they introduced this CCT rejection. 

https://github.com/home-assistant/core/pull/64581

PS: I'm experiencing this issue with an SK6812 RGBW strip, a WLED reset didn't solve it.

blazoncek, 2023-02-27:
SK6812 is not a CCT strip. You may have enabled color correction. See the above link and disable color correction.

mrosseel, 2023-02-28:
> SK6812 is not a CCT strip. You may have enabled color correction. See the above link and disable color correction.

I did indeed enable 'White Balance Correction', after correcting the mistakes in my led config and disabled the correction, HA added the LED strip.

AstonishingAston, 2023-08-11:
Just in case anyone else like myself come across this thread in hope of finding a solution, I have found a workaround.
I've got some Athon WLED controllers that do both digital and analogue ribbons and was getting the same CCT error when trying to add to HA.

Ticking "Calculate CCT from RGB:" and making sure "White Balance correction:" is un-ticked then rebooting the WLED device had HA detect it instantly.

Might be a few options to fine tune the above (haven't tested these settings with any LEDs yet).

DjordjeMandic, 2024-03-21:
> seems like HA cannot determine whether it's RGB+CCT or CCT which is why they disabled all CCT handling to prevent unexpected behaviour. See PR below where they introduced this CCT rejection. 
> 
> https://github.com/home-assistant/core/pull/64581

Do you have any idea why it can't be determined? Is it because wled does not provide enough specific info or because HA just didn't implement it?

blazoncek, 2024-03-21:
It is the complexity of possible WLED configurations that gives HA headaches. Not the lack of API.

DjordjeMandic, 2024-03-22:
So I want to have in my living room sk6812 RGBW + sk6812 WWA and one normal CW/WW CCT strip. 

Should then I make 2 controllers (instances of wled)? One with CCT and one without CCT. Add one without CCT to home assistant and have full native support and find a way for another one with cct to add it to homeassistant in some way.

As I understood it, CCT on wled is there to allow input command of color tempearture and wled then calculates how much should it drive each channel of white color to get desired color temperature.

I have 2 options for now:
  - Use one hardware custom built wled controller with cct disabled: 
      - sk6812 rgbw (for animations)
         - full native HA support
      - sk6812 wwa (for white cct animations)
         - full native HA support, treated as RGB for now. HA could do some calculations and send RGB values 
      - pwm 2ch CW WW strip
         - only reason for this is beacuse one above is not supported in wled for its purpose yet.
         - would use it as 2 independent channels on HA and HA could do some calculations and send 2ch values
   - Use 2 hardware custom build wled controllers:
      - One with CCT disabled:
         - sk6812 rgbw (for animations):
            - full native HA support
         - sk6812 wwa (for white cct animations):
            -  full native HA support beacuse its treated as RGB for now. HA could do some calculations and send RGB values 
      - One with CCT enabled:
         - pwm 2ch CW WW strip
         - would find a way to control it via HA by sending CCT value from ha direclty.
         
Ideal option:
   - Use one hardware custom built wled controller with cct enabled:
      - sk6812 rgbw (for animations)
         - full native HA support, NO CCT needed for this
      - sk6812 wwa (for white cct animations)
         - full native HA support, full native WLED support, CCT enabled, controlled natively from HA.

Ideal option with workaround until its all impelmented in HA and WLED:
   - Use one hardware custom built wled controller with cct disabled:
      - sk6812 rgbw (for animations)
         - full native HA support
      - sk6812 wwa (for white cct animations)
         - full native HA support
         - implement some kind of my own CCT control wihtout animations until its impelmented by wled and ha
         
Can CCT be enabled and disabled per strip or its global config?

With which option should I go?

blazoncek, 2024-03-23:
For RGBW + CCT support talk to HA developers.

DjordjeMandic, 2024-07-22:
This might get fixed in September HA release.

closed 2024-11-26 by github-actions[bot]