wled/WLED #3013: Settings not saved on ESP01-1M

hphde, 2023-01-10:
### What happened?

I tried ESP Fastled first on my EPS01-1M with ESPhome fastled which worked fine except for the limitations then decided to switch to WLED.
Flash was done using https://install.wled.me/ and setting Wi-Fi was a success. Unfortunately after reboot/poweroff the settings are lost. Pressing the backup preset or configuration button gives a 404 Not Found on the requests, visible using debug tools but with not representation on the UI.

### To Reproduce Bug

Flashed 0.13.1, 0.13.3, 0.14.0-b1 on a ESP01-1M (black board) - always the same effect

### Expected Behavior

settings saved

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
n/a
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

softhack007, 2023-01-10:
Hi,
A few questions to better understand what is the root casuse of your problem:

* Please add a screenshot of the WLED "info" page. What size do you see for "Filesystem"? 

> visible using debug tools
* What debug tools are you using?
* please add your "debug tools output" to the problem report

hphde, 2023-01-11:
This is from the failing instance:
[image]

As soon as I saw the filesystem at 100% I downloaded the ESP01.bin and flashed this to another device:
[image]
This one has no issues and everything works as expected.

So I am assuming the device detection is not working as expected on the install website and flashes something else?

blazoncek, 2023-01-11:
You need to flash 1M firmware version for ESP01, except if you know you have 4M flash chip on ESP01 (a rarity).

hphde, 2023-01-11:
I have changed the topic to reflect the issue found. So it's not an issue with the firmware itself but with https://install.wled.me/

blazoncek, 2023-01-11:
Than please close the issue here.

Aircoookie, 2023-01-11:
install.wled.me always installs the 4M binary, ESP Web Tools does not yet distinguish between flash sizes, so we can't serve the appropriate binary. If we can get ESP Web tools to add support for flash size detection, it would be great for all non-4M-flash boards: https://github.com/esphome/esp-web-tools/issues/111

Aircoookie, 2023-01-11:
Closing in favor of #2367 , which is the same core issue, just a different symptom.

closed 2023-01-11 by Aircoookie

hphde, 2023-01-11:
I suggest to point this 4m out on the web installer because it is not obvious