wled/WLED #3016: WLED library for use in mobile applications

mohamadhadibi, 2023-01-11:
Hi, I have a mobile application in Flutter that I want to use WLED in it as a library. I want to sync some audio and video with some WLED priests. Any hints will be appreciated.

blazoncek, 2023-01-11:
WLED is not a library. It is an ESP8266 or ESP32 firmware to drive LEDs.
There is no way you can stream audio to WLED in any current implementation/fork.
You will need to implement audio streaming yourself.

For *video* there is a DDP/E1.31/ArtNet interface built into WLED. Video is considered in a very loose terms.

softhack007, 2023-01-11:
> I want to use WLED in it as a library

WLED is not available as a component / library, but - as @blazoncek explains - WLED support standard protocols that might be available in external 3rd party libraries. See https://kno.wled.ge/interfaces/e1.31-dmx/ for an overview of protocols you can use.

Additionally, there is this library created by a user, that communicates using the audioreactive "[UDP sound sync](https://github.com/atuline/WLED/wiki/UDP-Sound-Sync)" protocol: https://github.com/netmindz/WLED-sync .

The "WLED APP" is actually just a wrapper (web view) that displays the web pages that are served by the esp32/esp8266 that run WLED. You can find the source code in this project: https://github.com/Aircoookie/WLED-App.

Closing this request, as there a no plans in the WLED core team to create additional libraries.

closed 2023-01-11 by softhack007

mohamadhadibi, 2023-01-12:
I know it is not a library, I want to make it in order to change it’s UI. I want to play some audio with my application (it’s completely software and I don’t need any hardware modules or stream library for it).
In my app I want to click on for example train button and I programmatically play train sound and simultaneously my LEDs simulate train station lights. I don’t want user see WLED app I want to play my train station light preset when I click on the button in my app. is there any way to do it now?
if not I want to implement it on my own, Please let me know what you can help me with.

blazoncek, 2023-01-12:
Use any API provided with WLED.
Use [KB](https://kno.wled.ge) to familiarize yourself with APIs.

softhack007, 2023-01-12:
If you just want some LED stripe effects as a library, then this might help: https://github.com/kitesurfer1404/WS2812FX