wled/WLED #3017: Analog LED PWM Duty cycle doesn't go beyond 50%

tsebaschtian, 2023-01-11:
### What happened?

I have connected an analog 12V LED via a N-Channel MOSFET to GPIO12 of an ESP8266.

Then I configured an Analog PWM-Strip in the LED-Preferences Tab.

Now i have a White channel slider in the UI, however, even if I put all sliders to the max, the LED never goes to full brightness.
So I tried to lower the PWM frequency, that didn't help.
So it hooked up an Osciloscope and found that the Duty Cycle never goes beyond roundabout 50%.

I'm unsure wether this is expected behaviour or just a setup issue.

### To Reproduce Bug

1. Fresh WLED Install
2. Set LED output to PWM
3. Duty cycle doesn't exeed 50%

### Expected Behavior

I would expect the Duty cycle to go from 0 to 100% (full on)

### Install Method

Binary from WLED.me

### What version of WLED?

0.13.3, 0.14.0

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Relevant log/trace output

```shell
{"rev":[1,0],"vid":2208222,"id":{"mdns":"wled-d77bde","name":"WLED","inv":"Light"},"nw":{"ins":[{"ssid":"Your_Network","pskl":0,"ip":[0,0,0,0],"gw":[0,0,0,0],"sn":[255,255,255,0]}]},"ap":{"ssid":"WLED-AP","pskl":8,"chan":1,"hide":0,"behav":0,"ip":[4,3,2,1]},"wifi":{"sleep":true},"hw":{"led":{"total":256,"maxpwr":850,"ledma":55,"cct":false,"cr":false,"cb":0,"fps":42,"rgbwm":3,"ins":[{"start":0,"len":1,"pin":[12],"order":1,"rev":false,"skip":0,"type":41,"ref":false},{"start":1,"len":255,"pin":[2],"order":0,"rev":false,"skip":0,"type":22,"ref":false}]},"com":[],"btn":{"max":2,"ins":[{"type":2,"pin":[0],"macros":[0,0,0]},{"type":0,"pin":[-1],"macros":[0,0,0]}],"tt":32,"mqtt":false},"ir":{"pin":-1,"type":0,"sel":true},"relay":{"pin":14,"rev":false},"baud":1152},"light":{"scale-bri":100,"pal-mode":0,"aseg":false,"gc":{"bri":1,"col":2.8},"tr":{"mode":true,"dur":7,"pal":0},"nl":{"mode":1,"dur":60,"tbri":0,"macro":0}},"def":{"ps":0,"on":true,"bri":128},"if":{"sync":{"port0":21324,"port1":65506,"recv":{"bri":true,"col":true,"fx":true,"grp":1,"seg":false,"sb":false},"send":{"dir":false,"btn":false,"va":false,"hue":true,"macro":false,"twice":false,"grp":1}},"nodes":{"list":true,"bcast":true},"live":{"en":true,"mso":false,"port":5568,"mc":false,"dmx":{"uni":1,"seqskip":false,"addr":1,"mode":4},"timeout":25,"maxbri":false,"no-gc":true,"offset":0},"va":{"alexa":false,"macros":[0,0]},"blynk":{"token":"","host":"blynk-cloud.com","port":80},"mqtt":{"en":false,"broker":"","port":1883,"user":"","pskl":0,"cid":"WLED-d77bde","topics":{"device":"wled/d77bde","group":"wled/all"}},"hue":{"en":false,"id":1,"iv":25,"recv":{"on":true,"bri":true,"col":true},"ip":[0,0,0,0]},"ntp":{"en":false,"host":"0.wled.pool.ntp.org","tz":0,"offset":0,"ampm":false,"ln":0,"lt":0}},"ol":{"clock":0,"cntdwn":false,"min":0,"max":29,"o12pix":0,"o5m":false,"osec":false},"timers":{"cntdwn":{"goal":[20,1,1,0,0,0],"macro":0},"ins":[]},"ota":{"lock":false,"lock-wifi":false,"pskl":7,"aota":true},"um":{}}
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

cad435, 2023-01-28:
I experience the same behaviour

cad435, 2023-01-28:
I temporarily fixed it by enabling "gamma correction for brighness" and setting the brightness-factor to 200%

Not sure if this is intentional, but it fixed it for me

isaacolsen94, 2023-12-04:
I'm also having this issue. Did anyone figure out what was going on?

sartorisjohn, 2023-12-17:
I am also seeing symptoms that led me here.  What I noticed was that while I had the white channel and the segment at max brightness I had the system brightness in the top right pretty low.  This is the first time I'm really trying to use WLED for a physical multi segment output.  I've made logical segments before, but the integration of multiple PWM seems to be a little rough.

[image]

I'm wondering about the capabilities of the software PWM and if we are noticing the affect of splitting the CPU time.  For first comment with pics @cad435 , with 2 full PWM channels I could see it being a split of time that the system can drive each pin.  I'd be curious if adding a third pin brought the duty cycle down again.  

ESP8266 only has a single chan of hardware PWM and I am not sure if/how WLED uses it.

ESP32 offers upto 16 chan of hardware pwm.  Again, not sure if/how WLED can use this.

blazoncek, 2024-02-26:
Anyone interested in helping solve this issue? I have a possible fix which also includes greater duty cycle resolution on ESP32.
There are no changes for ESP8266 though.

If you mix analog and digital LEDs please uncheck (disable) automatic brightness limiter (ABL). ABL may be responsible for lowering output brightness, even for analog LEDs, in all versions prior to 0.15. In version 0.15 ABL will not affect PWM outputs.

DedeHai, 2024-02-26:
I can not recreate this problem on a ESP8266 D1 mini. Compiled from Main Branch (0.14.1 ), factory reset, output on GPIO12, PWM White. Output goes from 0 to 100%

blazoncek, 2024-02-27:
Verified with beta testers with access to oscilloscope and all versions of 0.14 and above work as expected.
If you mix digital and analog LEDs, ABL will reduce maximum brightness on analog as well with versions <0.15.
Closing

closed 2024-02-27 by blazoncek