wled/WLED #3018: Add option to reset segments and LED config

vguttmann, 2023-01-11:
**Is your feature request related to a problem? Please describe.**
Home Assistant has a problem with WLED if the segments don't start from zero, and in some other cases, I was also getting other errors that stopped the browser interface from working

**Describe the solution you'd like**
A way to reset everything related to LEDs without touching the WiFi config

**Describe alternatives you've considered**
Reflashing the device, but then I'd have to desolder it and everything 

**Additional context**
This is also partly home assistants fault, but having the option to recover from a horribly borked LED config would still be good.

Thank you for your ideas for making WLED better!

blazoncek, 2023-01-11:
I don't fully understand what are you trying to describe. So I'll do my best...
*Factory reset* is available in Security settings. You can also *reset* segments to a single default segment by clicking UI element. Segments are a variable thing, one preset can have only one segment while another preset can have as many as 32 segments. If HA can't cope with that you should talk to HA team.

dosipod, 2023-02-08:
Not sure about HA issues as there are many . The solutions needed is **" A way to reset everything related to LEDs without touching the WiFi config "**   . Inst that  already there if you hard code your wifi and ssid  in my_config.h

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek