wled/WLED #3021: External Relay toggling during boot-up of WLED

Nippey, 2023-01-12:
### What happened?

Hi,
I have an external relay connected to GPIO13 (GPIO12 did not work due t boot issues)

Relay is low-active.
Relay has an internal pull-up.
WLED is configured to default OFF on boot,

So, on power-on, WLED should pull GPIO13 UP and stay there until I turn the LEDs on.

Unfortunately, during boot, the Relay does "Click-Clack", turning attached light ON-OFF.

Can I prevent this behaviour?

### To Reproduce Bug

- Define Relay Pin (I use GPIO13)
- Define default OFF.
- Connect Pull-Up to GPIO13.
- See GPIO13 shortly pull LOW during boot.

### Expected Behavior

I would expect GPIO13 to directly go to conifgured state OFF, if also LEDs are configured as default OFF.

### Install Method

Binary from WLED.me

### What version of WLED?

WLEDMM_0.14.0-b1.14 esp32_4MB_max.bin (2301083)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
ESP-WROOM-32 / ESP32-D0WDQ6
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Nippey, 2023-01-12:
Thanks for the comment!
I tried to implement the default-state of the relay, in case the ESP is still in reset.

So, default direction for WLED is LOW, is assume?
Then I will just rewire (relay has three terminals NC and NO) and change the internal pull-p to an external pull-down.

Thanks
BR
Nippey

closed 2023-01-12 by Nippey

blazoncek, 2023-01-12:
Default is what Espressif set it to. Some GPIOs will output random data.
Check Espressif documentation.

Nippey, 2023-01-12:
Aah, I understand, I may even solve it by using yet another pin that doesn't serve further espressif-defined functions.
I did not think that far :)