wled/WLED #3027: HC-SR04 false readings

Alfagek, 2023-01-14:
### What happened?

With the usermod: Staircase

I really don’t know any more we’re I can find the problem. It has to be somewhere in the FW. Fighting with it for 5 weeks now, changed data cables to shielded cables, using 10k pulldown resistor etc etc

Problem is that the ultrasoon sensors is getting triggered at moments there is not a reason for. Setting detection 75cm, and the distance between sensor and wall is between 81-84. Also if I set detection distance on 80cm in Wled, it’s in real about 40cm before its triggered if I use my hand. If i set it on 105cm its about 10cm from the wall. But still getting false detections from the sensor 

Using a esp32dev board (esp32 wroom)
Top trigger: gpio25
Top echo: gpio26
Bottom trigger: gpio14
Bottom echo: gpio27

If get a second esp32dev board and installing the 2 ultrasoon sensors with esphome on the same pins and setup, I get a nice reading. Nice between 81-84cm.

Also did get a brand new esp32mini, flashed with esp flasher from Srg74 and compiled version from his github (version 13.3 and version 14.0-b1)

Also i did noticed that when i use a pir sensor (hw416), it doesn't want to do any detection. Including trying with a pulldown resistor with the usermod.

### To Reproduce Bug

Al info all ready type above

### Expected Behavior

Normal reading distance

### Install Method

Self-Compiled

### What version of WLED?

0.13.3 (2208222)

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-15:
Ultrasonic sensor may be quite inaccurate in combination with WLED.
You may want to modify usermod code to incorporate interrupts in case US sensor is used.

blazoncek, 2023-01-18:
I am talking about WLED doing multitude things during waiting for sound bounce.
So the timing may get skewed.

I would recommend using PIR sensors instead.

Alfagek, 2023-01-22:
Thanks @blazoncek. When i have again some energy, i wil maybe try the pir sensors. 
Working now with ESPHOME and the ultrasonic sensors beside the wled with usermod.

closed 2023-01-22 by Alfagek