wled/WLED #3028: Even rows not uploaded correctly in 16x16 2D panel via JSON

werkstrom, 2023-01-14:
### What happened?

When uploading a 16-16 setting only odd rows (1, 3, 5, ...) are handled correctly. Even rows are ignored.

[image]

Desired result to the right incl. console validating the request succeeds.
Actual result in WLED peek to the left.

*Have also verified by sending the JSON using NodeRED, same result
*Have tested both HEX/DEC and Single/Range
*Have tried splitting into two separate commands (and olso three)
*Have tested on two separate D1 minis
*Have verified number of colors sent are 256 as it should (plus the extra "dummy" as per earlier issue.

Only via peek, have no actual LED to test with.

[image]

[image]

### To Reproduce Bug

Send this JSON:
~~~
{
  "on": true,
  "bri": 127,
  "seg": {
    "i": [
      0,
      'dummy',
      '106c21',
      '106c21',
      '093c12',
      '010602',
      '010602',
      '010602',
      '0e601d',
      '106c21',
      '106c21',
      '0f661f',
      '010602',
      '010602',
      '010602',
      '083510',
      '106c21',
      '106c21',
      '106c21',
      '093d13',
      '3d0d04',
      'a30000',
      'a20000',
      'a30000',
      '081c08',
      '0f671f',
      '106c21',
      '041e09',
      'a30000',
      'a30000',
      '740000',
      '0d0b03',
      '083911',
      '106c21',
      '0b4c17',
      '350401',
      'c51f1f',
      'ff4747',
      'ff4c4c',
      'ff0000',
      '820000',
      '09310f',
      '083711',
      '770000',
      'ff0000',
      'ff0000',
      'f70000',
      '760000',
      '1c0000',
      '0b4c17',
      '000000',
      'd00000',
      'ff9696',
      'ffffff',
      'ffecec',
      'ff0101',
      'ff0000',
      '4b0000',
      '390000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'f20000',
      '640000',
      '000000',
      '010401',
      'bd0000',
      'ffc8c8',
      'ffffff',
      'ff1111',
      'ff0000',
      'ff0000',
      'f30000',
      'f10000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'e20000',
      '590000',
      '010401',
      '010602',
      'ba0000',
      'ff2424',
      'fe2626',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'df0000',
      '580000',
      '010602',
      '000000',
      'c80000',
      'ff2727',
      'fe3535',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ed0000',
      '600000',
      '000000',
      '020b03',
      'af0100',
      'fabcbc',
      'fff7f7',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'fe0000',
      'd80000',
      '560000',
      '020b03',
      '0f661f',
      '0b0e04',
      'a20c0c',
      'ff1010',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'e30000',
      '4d0000',
      '060702',
      '0f661f',
      '106c21',
      '0c4e18',
      '241406',
      '9c0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'e40000',
      '600000',
      '0a1005',
      '0b4b17',
      '106c21',
      '106c21',
      '106c21',
      '0b4a17',
      '131005',
      'c20000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ff0000',
      'e30000',
      '730000',
      '020e04',
      '0a4515',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '0c5219',
      '030100',
      'f70000',
      'ff0000',
      'ff0000',
      'ff0000',
      'ea0000',
      '750000',
      '000100',
      '0b4b17',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106a20',
      '0c5319',
      '160000',
      'eb0000',
      'ff0000',
      'f70000',
      '790000',
      '0a0000',
      '0b4e18',
      '106920',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '0a4214',
      '470000',
      'b40000',
      'ab0000',
      '180000',
      '0a4114',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '083310',
      '2e0100',
      '3b0000',
      '072e0e',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '05200a',
      '041807',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21',
      '106c21'
    ]
  }
}
~~~

### Expected Behavior

Should show the image to the right, not the current one to the left.

### Install Method

Binary from WLED.me

### What version of WLED?

v0.14.0-b1 "Hoshi"

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-01-15:
First , we are testing all below on esp32 so might be that is something to testing first  ..
[image]

The local version worked out of the box on the esp32 .  Files below are that we used for the test , please test that and post the same pics so we could compare 
[Retro Star.zip](https://github.com/Aircoookie/WLED/files/10419774/Retro.Star.zip)

werkstrom, 2023-01-15:
Confirmed fixed in latest nightly, 2301140. closing.

werkstrom, 2023-01-15:
unfortunately bug has returned. Now in latest beta as well as b1.
[image]
Can it be that something gets corrupt? I could take some sort of log dump if I can get instructions. Problem persists after reboot.

werkstrom, 2023-01-15:
Appears it returns after some time. Noticed issues after changing led/2D setup...Reboot fixed the issue

dosipod, 2023-01-15:
Are you are seeing that on esp32 ?  We did not setup any esp8266 to  be used with a matrix and 2d to avoid the possible heartache , if you still would want the end goal to run that on esp8266 then we could build one with esp8266 to test with you . But might give us trouble on the long run but we can still do it to test

werkstrom, 2023-01-16:
Waiting for ESP32s to be shipped. Only have my two D1 mini 8266. So, If you have no issue, I guess your conclusion on them not being up to the task, appears correct.

werkstrom, 2023-01-16:
I also have no LED matrix to try on so I see the issues in peek, so the issue can be with peek as well...

blazoncek, 2023-01-16:
Peek is consistent with LED output.

werkstrom, 2023-01-18:
Ok, I would say evidence strongly suggest this happening when rearranging led/panel setup. Happened again after flawless operation when I changed setup from 1 32x32 --> 2 16x16 panels
[image]

blazoncek, 2023-01-18:
Did you change segments? After changing 2D set-up it may be necessary to update segments.

werkstrom, 2023-01-18:
Did it before with only one segment, this time (the grey image) is with two segments, that I believe I updated.

blazoncek, 2023-01-18:
Use debug build and monitor serial output. Dump matrix ledmap after 2D change.

werkstrom, 2023-01-22:
It appears matrix layout is not completely updated until reboot, under at least some circumstances. Noticed that the JSON output does not update until reboot. 
Before reboot
[image]
After reboot
[image]

This is consistent with how peek behaves, so that would indicate something underlying not being updated.

Still only have my D1-minis 8266 to run on.

blazoncek, 2023-01-22:
The data you post is from **segment** not matrix.
Updat your segment to reflect new matrix set up.

Unfortunately, updating segments after matrix configuration is not yet implemented.

werkstrom, 2023-01-22:
OK, makes sense if the segment falls outside of the matrix then because in my case going from 32->16 pixels would make the segment think there are pixels available to plot led 17-32 to the right of the panel. Rendering every even row out of view 😆... Which is exactly the result we're seeing. That would also account for some very funky results I've seen before, say, going from say 24 -> 16 giving you a very messed up image. Could I perhaps propose some sort of check/warning that segments are set up that falls outside the available matrix? Since that should hardly be desirable?

blazoncek, 2023-01-22:
We already have *automatic segment creation* as part of multi strip/pin set up.
Unfortunately multiple *strips/pins* can all be part of a single matrix so that logic is out of scope.

We may add a logic to detect if a 2D segment (only one) equals previous 2D matrix set up and update it after the change.
You can achieve the same by adding one *phantom* segment and then resetting all segments.

blazoncek, 2023-02-01:
Resolved by resetting segments on 2D change (not best solution, but it works to alleviate the problem).

closed 2023-02-01 by blazoncek