wled/WLED #3029: WLED on ESP32- device always rebootin

JimlyTheReal, 2023-01-14:
### What happened?

so far I always used ESP8266 units, like Wemos-D1 mini - problem free. 

Now I wanted to use WLED with an ESP32.
- No matter which binary (0.13.1, .2, .3, 0.13.3 sr, 0.14.0-b1) I install 
- no matter how, whether via the web installer or esptool, 
- no matter whether with or without bootloader
- no matter whether it is connectet via USB or an 5V power supply 
- no matter whether it is connectet to an LED string or not
After approx. 60s the ESP32-WROOM-32 (board with 32 PIN ESP32 DEVKIT V1) restarts.

No idea why!

### To Reproduce Bug

connect the flashed ESP to Power - even without connected LED strips

### Expected Behavior

well what, no reboot, just work

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1 any 0.13.x /+sr

### Which microcontroller/board are you seeing the problem on?

ESP32

### Relevant log/trace output

```shell
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401670d8  PS      : 0x00060c30  A0      : 0x80167abe  A1      : 0x3ffd7740
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x0000005a  A5      : 0x3ffd7bf4
A6      : 0x3ffdd821  A7      : 0x3ffdd87b  A8      : 0x80166bf2  A9      : 0x3ffd7720
A10     : 0x00000001  A11     : 0x3ffdd821  A12     : 0x0000005a  A13     : 0x00000001
A14     : 0x00060823  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x401670d5:0x3ffd7740 0x40167abb:0x3ffd77c0 0x40167b1b:0x3ffd77f0 0x4008b6ae:0x3ffd7820

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x401670d8  PS      : 0x00060a30  A0      : 0x80167abe  A1      : 0x3ffd7750
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x0000005a  A5      : 0x3ffdd228
A6      : 0x3ffdd90d  A7      : 0x3ffdd967  A8      : 0x80166bf2  A9      : 0x3ffd7730
A10     : 0x00000001  A11     : 0x3ffdd90d  A12     : 0x0000005a  A13     : 0x00000001
A14     : 0x00060623  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x401670d5:0x3ffd7750 0x40167abb:0x3ffd77d0 0x40167b1b:0x3ffd7800 0x4008b6ae:0x3ffd7830

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DOUT, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Ada

... and so on ...
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

JimlyTheReal, 2023-01-14:
I also develop in micropython and cpp - I don't have this problem there. So it must have something to do with the code?

blazoncek, 2023-01-15:
Use debug build and install exception decoder to PIO.
Then return back with another crash dump.

JimlyTheReal, 2023-01-15:
Hi thnx for ur response. I've compiled it as esp32dev with the build_flags = ${common.build_flags_esp32} -D WLED_RELEASE_NAME=ESP32 -D WLED_DISABLE_BLYNK #-D WLED_DISABLE_BROWNOUT_DET
  -D WLED_DEBUG

---DEBUG INFO---
Runtime: 30000
Unix time: 30,000
Free heap: 205712
Wifi state: 6
State time: 30000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 2455
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B
JSON buffer locked. (17)
JSON buffer size: 25 for request: 7
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 25 for request: 7
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 25 for request: 7
JSON buffer released. (17)
JSON buffer locked. (17)
JSON buffer size: 527 for request: 7
JSON buffer released. (17)
Forcing reconnect.
Connecting to Ghostbuster...
Writing settings to /wsec.json...
JSON buffer locked. (4)
JSON buffer released. (4)
Writing settings to /cfg.json...
JSON buffer locked. (2)
Not-Found HTTP call:
URI: /hotspot-detect.html
Captive portal
FileRead: /index.htm
JSON buffer released. (2)
Local time: 00:01
---DEBUG INFO---
Runtime: 60013
Unix time: 60,013
Free heap: 204908
Wifi state: 6
State time: 30000
NTP last sync: 999000000
Client IP: 255.255.255.255
Loops/sec: 2010
UM time[ms]: 0/1
Strip time[ms]: 0/2
Segments: 1 -> 72B
Modes: 4*187=748B
Data: 4*187=748B
Map: 2*0=0B

Connected! IP address: 192.168.2.58
Init STA interfaces
mDNS started
Access point disabled (handle).
FileRead: /index.htm
Not-Found HTTP call:
URI: /skin.css
FileRead: /skin.css
JSON buffer locked. (17)
JSON buffer size: 7262 for request: 6
JSON buffer released. (17)
[... 215 lines cut ...]
JSON buffer size: 2057 for WS request (1127).
Sending WS data to a single client.
JSON buffer released. (12)
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
Core 0 register dump:
PC      : 0x40168790  PS      : 0x00060a30  A0      : 0x80169176  A1      : 0x3ffd7530  
A2      : 0x00000000  A3      : 0x00000010  A4      : 0x0000005a  A5      : 0x3ffdd044
A6      : 0x3ffde185  A7      : 0x3ffde1df  A8      : 0x801682aa  A9      : 0x3ffd7510  
A10     : 0x00000001  A11     : 0x3ffde185  A12     : 0x0000005a  A13     : 0x00000001
A14     : 0x00060623  A15     : 0x00000010  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000

ELF file SHA256: 0000000000000000

Backtrace: 0x4016878d:0x3ffd7530 0x40169173:0x3ffd75b0 0x401691d3:0x3ffd75e0 0x4008b6ae:0x3ffd7610
  #0  0x4016878d:0x3ffd7530 in mdns_parse_packet at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:4543
  #1  0x40169173:0x3ffd75b0 in _mdns_execute_action at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:4543
  #2  0x401691d3:0x3ffd75e0 in _mdns_service_task at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/mdns/mdns.c:4543
  #3  0x4008b6ae:0x3ffd7610 in vPortTaskWrapper at /home/cschwinne/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c:355 (discriminator 1)   

Rebooting...

and so on...

- these information are enough?

blazoncek, 2023-01-15:
Something in mDNS is causing null pointer to be used.
It may be something in your network. Can you check what is happening with WireShark?

Lollo79, 2023-01-15:
FritzOS 7.50?

JimlyTheReal, 2023-01-15:
Wow, ich changed the connection to my guest wifi - and it working now without rebooting....
Yes i use a FritzBox with OS 7.50 - are there any issues kown? (especially only with the ESP32?)

JimlyTheReal, 2023-01-15:
thnx blazoncek for right the direction!

blazoncek, 2023-01-15:
Closing as a duplicate of #2932

closed 2023-01-15 by blazoncek