wled/WLED #3030: Usermode Temperature

EDSTOBI, 2023-01-15:
### What happened?

If the temperature sensor is not detected during boot, this usermod is deactivated

### To Reproduce Bug

o which GPIO must the sensor be connected so that it is detected, or where can I define this GPIO in the code?

### Expected Behavior

I have configured an esp32 WT32-eth with temperature sensor and do not figure out wher to connect the sensore

### Install Method

Self-Compiled

### What version of WLED?

2301140

### Which microcontroller/board are you seeing the problem on?

ESP32, Other

### Relevant log/trace output

```shell
No usermode with temperature sensor
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-15:
usermods settings page.
it is not an issue with WLED as Temperature usermod is working well with or without Ethernet.

EDSTOBI, 2023-01-15:
Hi
usermods settings page dos not show Temperature.
Reademe says: If the temperature sensor is not detected during boot, this usermod is deactivated.
What does this mean? not changeable or not visible

blazoncek, 2023-01-15:
Then you do not have temperature usermod compiled in.
Use a binary with temperature usermod compiled in.

closed 2023-01-15 by blazoncek

EDSTOBI, 2023-01-15:
Yes :-(
I changed in platformio.ini 

# Release / CI binaries
default_envs = esp32_eth

 #For Dallas sensor uncomment following 2 lines
    OneWire@~2.3.5
    milesburton/DallasTemperature@^3.9.0

Compiling with now error, WLED is up and running.
What do I miss?

softhack007, 2023-01-15:
> What does this mean? not changeable or not visible

If you have a firmware with usermod temperature compiled in, then the usermod setting will always be visible, and changeable. "deactivated" mean not active, I.e. not reading or reporting data from the sensor as it was not found.

About your question on source  code:
https://github.com/Aircoookie/WLED/blob/1f135f1fa5f656db3e851b05905825d0a7fb74a4/usermods/Temperature/usermod_temperature.h#L6-L13

https://github.com/Aircoookie/WLED/blob/1f135f1fa5f656db3e851b05905825d0a7fb74a4/usermods/Temperature/usermod_temperature.h#L337-L342

blazoncek, 2023-01-15:
You need to add ` -D USERMOD_DALLASTEMPERATURE`

softhack007, 2023-01-15:
> What do I miss?

Please join discord to get help and support from the WLED community. A bug report is not the right way for asking help with compiling usermods.

Join the Discord server to discuss everything about WLED!

[image]

Check out the WLED [Discourse forum](https://wled.discourse.group)!