wled/WLED #3032: [Feature Request] Add Rotation to 2D effects

mermelmadness, 2023-01-16:
**Is your feature request related to a problem? Please describe.**
For making cube devices, or if a user wants a diamond shape for their matrix

**Describe the solution you'd like**
A slider allowing rotation in degrees.  -/+ 180 (or 0-360)

**Describe alternatives you've considered**
Manually making API calls, but this only works for static pixels

Thank you for your ideas for making WLED better!

blazoncek, 2023-01-22:
What would be a use case for that?
Offset (phase/rotation as you describe it) was removed for 2D functionality.

mermelmadness, 2023-01-23:
I'm building an 8x8x8 cube.  The orientation would be so that the left and right visible panel would be head on, with the top panel also visible and it would be below my eye line.  Ideally, I'd like that top panel to be orientated at 45 degrees to be head on with my vision.

blazoncek, 2023-01-23:
I am not seeing how that ("rotation") would benefit what you describe.

softhack007, 2023-01-23:
In current WLED, you can already achieve rotations in steps of 90 degrees, by using combinations of "flip horizontal" and "flip vertical" in segment settings.

Arbitrary rotation created some addition technical problems:
* speed of the rotation operation  - maybe a rotation matrix and fixedpoint math will help
* accuracy of the rotation - in integer coordinates, arbitrary rotation will always leave "holes" behind
* size - rotation  (example 45 degress) will leave blank area at the corners, because effects do not calculate normally-invisible parts. So either always render a 50% bigger area in effects, or find a way to scale the rendered pixels to the new viewport area--> both solutions cause performance loss, scaling causes blurring 

As far as I know, WLED currently is not ready to address these problems.

mermelmadness, 2023-01-23:
@softhack007 I wasn't talking about an animated rotation, more a static one to which to orient effects, but I think you're right about the geometry.  I had forgotten that it's a fixed number of pixels available, so that's not really doable.  Ok, I withdraw my request.

closed 2023-01-23 by mermelmadness