wled/WLED #3039: Read ESP Pins

maxpd1, 2023-01-18:
**Describe wanted Enhancement !**

I want to read out the other PINs on the ESP2866 (Wemos D1 Mini) even if wled firmware is installed.
iobroker offers a great adapter to see a lot information which WLED provides. Nevertheless he single PINs of the board are missing.

For example: connected with LED Stripe there is at the same time an HC-SR501 motion sensor. Via iobroker or other smart home software, I can detect the motion when the pin shows "true" state and control single segments.

**Why should we put effort in it ?**

You don't need a separate ESP2866 for motion sensor anymore.

blazoncek, 2023-01-18:
Use MQTT for this purpose.

closed 2023-01-18 by blazoncek

DutchmanNL, 2023-01-23:
@Aircoookie why should it Not be implementeren in websocket API? It Males No Sense to use 2 different protocols?

Aircoookie, 2023-01-23:
I don't t believe MQTT makes intrinsically more sense for this functionality than JSON (as websockets also support a non-polling publish if the pin state changes).

However, I don't think that it is a feature that is required in most installs and worth the JSON overhead by default. WLED can in principle offer the flexibility of ESPHome or the likes, but that is what usermods are for.
So I'd propose adding the functionality of reading and outputting GPIO values via `/json/info` as a usermod.

blazoncek, 2023-01-23:
I understood the question in relation to buttons and relay that are controlled/used by WLED.
As such, buttons are handled using MQTT so no additional programming is required and does not use resources if not requested. Websocket publishing will only be available if WLED state changes, not when button is pressed.

Although it is possible to write a usermod that can *override* button behavior and add JSON state/info publishing it is too cumbersome if only publishing is required.