wled/WLED #3040: Support WPA3 on Wifi

schildbach, 2023-01-18:
**Is your feature request related to a problem? Please describe.**
WLED 0.13.x on an ESP32 (Athom Addressable Strip Controller) connects to Wifi only via WPA2. With WPA2, several kind of attacks are possible that are not possible with WPA3.

**Describe the solution you'd like**
Support WPA3. The ESP32 SDK seems to be able to do this out of the box. Tasmota has enabled this too.

**Describe alternatives you've considered**
If WPA3 cannot be considered, supporting PMF (protected management frames) on WPA2 would provide a similar level of security.

blazoncek, 2023-01-18:
Can you point to the documentation on this?
Especially for Arduino platform.

schildbach, 2023-01-18:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi-security.html

schildbach, 2023-01-18:
https://www.arduino.cc/reference/en/libraries/wifi/ only mentions WPA2. I hope this doesn't mean Arduino doesn't support WPA3.

schildbach, 2023-01-18:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi-security.html#wpa3-personal

schildbach, 2023-01-18:
https://www.arduino.cc/reference/en/libraries/wifi/ only mentions WPA2 and below. I hope this doesn't mean Arduino doesn't support WPA3.

softhack007, 2023-01-19:
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wifi-security.html

The first mention of WPA3 security mode is in the IDF-v4.3.x docs. So it might be that WPA3 is only supported when building with the newer platform framework - WLED is still on ESP-IDF 3.3.6 (`platform = espressif32@3.5.0`) while you might need `platform = espressif32@ ^5.0.0` or later to have the WPA3 functionality included in the framework.

Currently we cannot simply "upgrade" the platform, due to crazy bugs and stability problems in ESP-IDF 4.x.y - WLED on esp32 would basicially become unusable and crash randomly. But OK, at least we could have WPA3 ;-) When espressif would finally get their act together and start testing/fixing regressions on esp32, then -maybe- WPA3 could come in future WLED release ...

Jason2866, 2023-01-25:
@softhack007 Latest Arduino core 2.0.6 is stable. Tasmota does run rock solid with.
Code needs to be adopted to the changed APIs ;-)

softhack007, 2023-01-25:
> @softhack007 Latest Arduino core 2.0.6 is stable.

As much as I want to belief you ... I've tried the latest core 2.0.6 on a -C3 and -S2, and what i saw was worse than 2.0.4/2.0.5. "Classic ESP32" also crashes too often, compared to the rock-solid build with platform version 3.5.0. Maybe we are not ready yet for new cores and newest ESP-IDF.

From my (limited) experiences, arduino-esp32 2.04/2.0.5  (and ESP-IDF 4.4.1 / 4.4.2) still seem to be the better choice when a new platform version is absolutely necessary.

Jason2866, 2023-01-25:
Just can say Tasmota runs weeks without an reboot or WDT on ESP32, ESP32-S2, ESP32-S3 and the ESP32-C3. Yes, it needs changes in code, to get there.
Don't get me wrong, in no way i mean the next statement offending.
WLED is and was (until now) never rock solid on an ESP32.
Never managed to have an uptime over 2 months without an reboot or WDT.
With the old core I doubt you will manage this. The underlying IDF and Arduino code has too many bugs.
To be clear i love WLED and use it!!!

blazoncek, 2023-01-25:
@Jason2866 could you help us with issues with LittleFS on regular ESP32 and Arduino core 3.5+?

Jason2866, 2023-01-25:
The issue with LittleFS and the older cores is related to the not correct builded precompiled  bootloaders. There is no real fix for. Only moving to actual core (2.0.5 and 2.0.6)
Or you have to backport the way the bootloader is builded introduced with core 2.0.5 to older cores. One of more reasons why i started to maintain our own fork and compile the Arduino libs our self. 
In short flash mode is not always correctly initialized (from the Bootloader) and data gets corrupted when flash is accessed from LittleFS. That's not LittleFS fault. 
Probably you have a message like `corrupted dir pair`

Jason2866, 2023-01-25:
So long story short. To avoid flash corruptions (not restricted to LittleFS!) cores before 2.0.5 can only be used in mode `DOUT`

blazoncek, 2023-01-25:
That is an eye opener. Thank you.
We experimented with QIO and DIO and flash speed of 80MHz due to occasional flickering while frequently accessing flash and/or PSRAM during effect calculations.

To summarize:
- update to most recent bootloader
- use DOUT flash mode
- user more recent PIO platform if it is still compatible with WLED components
- try alternative flashing with esptool

Is that right?

Jason2866, 2023-01-25:
Yes, right direction.

MacMika, 2024-12-28:
same here. Are there any plans?

Jason2866, 2024-12-28:
When WLED is moving to actual Arduino core 3.x. there will be WPA3 support.
As long this is not done there is no WPA3.

schildbach, 2025-04-30:
It seems version 0.14.4 has WPA3, at least the version that on the Athom/IoTorero ESP32C3 controller, arriving by mail today.

If that's true, many thanks for implementing WPA3!

schildbach, 2025-05-01:
Version 0.15.1.beta2 on Athom ESP32 "Music Addressable LED Strip Controller" still connects via WPA2.

I guess I will update the ESP32C3 to stock WLED too, to check if it retains its WPA3 capability.

blazoncek, 2025-05-01:
WLED does not enforce or *enable* certain WiFi protocols, it uses whatever framework supports.
ESP32 uses ancient framework.

schildbach, 2025-05-01:
So ESP32 uses a framework (Arduino core, I assume) version different from ESP32-C3? If so, where can I see the Arduino version used? Is that perhaps something that could be added to the "Info" screen?

blazoncek, 2025-05-01:
It already is.

martijnbuts, 2026-02-26:
I have wled version 15.3 with environment version v3.3.6. Is there any option to get WPA3 working?

netmindz, 2026-02-27:
You don't say which board you are using @martijnbuts but if you look in the info page and see if the platform version is 3 or 4

If it's still 3, please try the esp32 V4 bin, that *might* have a support

netmindz, 2026-03-08:
@coderabbitai what version of the ESP IDF is required to support WPA3 ?

schildbach, 2026-03-08:
@netmindz My home router (a FritzBox, very common in Germany) doesn't have "WPA3 only". Rather, the highest setting is "WPA2 + WPA3".

Yet, WLED is the only firmware that still connects via WPA2 although it should be possible to use WPA3 on that hardware. Is it perhaps possible to restrict WLED to WPA3, or at least assign a higher priority to WPA3?

Jason2866, 2026-03-08:
@schildbach There is no priority setting in IDF core. Would have to be implemented manually. Not recommended since this will add complexity and additional time for connect.
Normally this is done in the AP. Bad luck Fritz Box does not provide this feature. Probably done not to generate issues from users not knowing what the do.

softhack007, 2026-03-08:
@Jason2866 would we need to add specific arduino-esp32 API calls for enabling WPA3 with the `WiFi` object? Similar to WPA2-enterprise which required some special handling in the WLED code?

Jason2866, 2026-03-08:
Looks like ;-)

schildbach, 2026-03-08:
Well, Tasmota priorizes features over security. So I dropped it from my home, in most cases in favor of ESPHome.

WLED on the other hand so far hasn't made an "official" statement about this. To me it appears WPA3 is supported, but a bug prevents it from working in all cases.