wled/WLED #3045: Possible Sync issue with 14b1

Olynthus, 2023-01-22:
### What happened?

When I updated from 13.3 to 14.b1 on all 5 of my ESP8266 cards the sync function stopped working. After going back to 13.3 all is good now.

### To Reproduce Bug

I have 5 ESP8266 cards. 4 are controlling strings of lights on my home. One of the ESP8266 cards just acts as a master to control all the lights with whatever I do on the master. The Master is just plugged on a power source via USB and that has all my Playlist effects on it. With 14.b1 that functionality stops working. The only actual effect that seems to sync is the brightness of the LEDs they do go brighter when I select a Playlist function on the Master ESP8266 instead of the dim white I have them all set to on power up. Is there a new setting I'm not aware of? They are all working now that I have gone back to 13.3 on all the ESP8266 cards. Thanks 

### Expected Behavior

I expected sync to work like it did with 13.3.

### Install Method

Binary from WLED.me

### What version of WLED?

14.b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-22:
I cannot reproduce.
I have 6 devices using 0.14 and they all sync.

Did you change sync groups or ports?

Olynthus, 2023-01-22:
Updated then tested, nothing else was done. Perhaps I need to go through all the setting tabs and save the configurations of each setting page with the new 14.b1 firmware?

softhack007, 2023-01-22:
It could help to do a full factory reset after update, then type in your settings again for each device. You could make screenshots in 0.13.3 so you get everything right.

foorschtbar, 2023-03-31:
Same probleme here. Two fresh installed v0.14.0-b1 and no sync working :(

blazoncek, 2023-09-18:
No response received, closing.

closed 2023-09-18 by blazoncek