wled/WLED #3046: Command on multi segment 2D not rendered correctly

werkstrom, 2023-01-22:
### What happened?

When sending commands to render an image on a segment of a multi segment setup the command (image) does not get rendered correctly. Several strange behaviors. Images gets rendered on the wrong segment, images gets rotated and mirrored. 

### To Reproduce Bug

Setup is a three 16x16 panel setup, meaning it takes up 3/4 of a 32x32 matrix. (Why would I have such a strange setup? Because it is the setup you'd need for a cube where you want to be able to address each panel individually)

Panel setup:
[image]

Segment setup:
[image] 
[image] 
[image]

From what I can tell, all segments are reported correctly in JSON req, 
`[{"id":0,"start":0,"stop":16,"startY":0,"stopY":16,"len":16,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":1,"start":16,"stop":32,"startY":0,"stopY":16,"len":16,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Top Right","col":[[255,170,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0},{"id":2,"start":0,"stop":16,"startY":16,"stopY":32,"len":16,"grp":1,"spc":0,"of":0,"on":true,"frz":false,"bri":255,"cct":127,"n":"Bottom Left","col":[[255,170,0],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"c1":128,"c2":128,"c3":16,"sel":true,"rev":false,"mi":false,"rY":false,"mY":false,"tp":false,"o1":false,"o2":false,"o3":false,"si":0,"m12":0}`

As is matrix (32x32)
`"matrix":{"w":32,"h":32}`

First off, this is what is rendered when opening peek and turning the lights on.
[image]
(should be a full 16x16 in the bottom left corner)

Then I send this image to the different segments:
[image]

First sending it to seg 0 it gets rendered as expected:
[image]

Sending it to segment 1 (should be top right quadrant) It gets rendered in segment 0's space.
[image]

It should be noted that if I sent the image to seg 1 before rebooting the image was rendered in the wrong plac but rotated and mirrored. After reboot it no longer does that. Also the segments were gone and had to be set up again.
[image]

And after a while of testing the device (d1mini) stops responding and I have to reboot. Would be great to get some help in doing more testing.

### Expected Behavior

Should render in correct segment

### Install Method

Binary from WLED.me

### What version of WLED?

2301140

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-22:
It renders for me correctly and segments behave as expected.
Please verify your LED count and 2D set up.

werkstrom, 2023-01-22:
Added to the initial description (so it is complete). I might be missing something, but it looks OK to me...

werkstrom, 2023-01-22:
[image]
Issue is consistant when using effects. This is after a fresh reboot, so it is not the "sending command through JSON" that triggers the behavior.

werkstrom, 2023-01-22:
Also consistent independent of led count/panel size
[image]

werkstrom, 2023-01-22:
Setting up 
A. LEDs for an entire square area (8x8x4 in this case)
B. Setting up panels to cover the entire area (4 panels 8x8)
C: Setting up segments for each panel

Renders correctly (and effects and uploads work as expected)

[image]

[image]

blazoncek, 2023-01-22:
Just a side note: ESP8266 may have memory issues if you want to have matrix larger than 16x16.

werkstrom, 2023-01-22:
Removing the lower right segment, works as expected.
[image]

Removing the "ghost" fourth panel also works (leds still defined as (8x8x4) JSON command and effects work.
[image]

Removing the 64 "ghost" LEDs from LED setup, reintroduces the issue

[image]

closed 2024-04-23 by github-actions[bot]