wled/WLED #3050: Scrolling Text flickers a lot if color intensity is not 100%

mermelmadness, 2023-01-24:
### What happened?

When using Scrolling Text effect on a matrix, the text flickers a lot at the same speed as the scroll effect.  This occurs when the top bar is anything other than 100%.  Brightness does not affect behavior.
[image]
This can be seen with scrolling or static text.
In addition, even if the text color is at 100%, adding any background color, even a very slight one, causes the same flicker.

https://user-images.githubusercontent.com/38080873/214208542-92cb9337-b4cb-4985-b146-c54b8ea147a8.mov

NOTE: This can be clearly seen using Peek.

### To Reproduce Bug

Make a Scrolling Text segment, set a color using the Default palette, then adjust the color intensity slider to witness the effect.

### Expected Behavior

With full color intensity the effect behaves as expected, with smooth scrolling or solid colors if static.  This should occur for all brightness and intensity levels.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b1 2301170

### Which microcontroller/board are you seeing the problem on?

ESP32

### Anything else?

This was mentioned in the `Issues` channel and also `beta-testing`
ALDIY confirms he sees the same issue.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

dosipod, 2023-01-24:
The same was noticed on multiple units  with  even the text is stationary , Just use random palette and you would see the same on leds and peek  . Default palette  works fine with us so might be a clue .
[image]
[image]

mermelmadness, 2023-01-24:
Default palette gives me flicker, although, setting the whole device brightness to very low and setting each segment brightness to half eliminates the flicker.
[image]
It's pretty dim for most applications, though.

dosipod, 2023-01-24:
Seems adjusting sliders speed or brightness could reduce it with random palette   , as said we did not notice flicker at all with default  palette . If you face issue wit default  palette   , please try to take video and show all the setting used  .  
[image]
This setting produced a solid blue colors text without any issue

mermelmadness, 2023-01-24:
All videos posted in initial issue are with Default palette.  Here's an additional comparison.
Overall brightness = 10
Each segment brightness = MAX

https://user-images.githubusercontent.com/38080873/214211443-a4564510-bd16-4edd-83ec-257f6f9c2899.mp4

Overall brightness = 10
Each segment brightness = Approx 30%

https://user-images.githubusercontent.com/38080873/214211630-df39f2b3-220f-4ba5-b22b-c08a07d1959b.mp4

WLED just crashed as I was taking the video...

blazoncek, 2023-01-24:
That is inherently caused by "trail" functionality.

mermelmadness, 2023-01-24:
In all examples, Trail is set to 0.

blazoncek, 2023-01-24:
Exactly!

blazoncek, 2023-01-24:
In case of static text (as in above examples) either use maximum trail length or use overlay mode (which may have other undesirable effects).

mermelmadness, 2023-01-24:
Setting Trail to 255 does not alleviate the issue.

I'm not sure how overlays work.  Can't seem to find any explanation on the KB.

blazoncek, 2023-01-24:
It does in my test cases.

mermelmadness, 2023-01-24:
https://user-images.githubusercontent.com/38080873/214330943-9392662e-b2b8-4c88-8b9b-33c60220c76e.mp4

[image]

blazoncek, 2023-01-24:
Use faster speed.

mermelmadness, 2023-01-24:
https://user-images.githubusercontent.com/38080873/214333095-25451a30-4985-4221-b40f-50a2e55dba55.mp4

[image]

Not sure if it comes through on video, but these settings it is less, but still very much there.  Overall brightness is 128, segment brightness is 100%.

mermelmadness, 2023-01-24:
It should be noted that if the Bg color is black, the flicker is not present.  Can Scrolling Text be paired with Solid using Overlays?

closed 2023-01-24 by blazoncek