wled/WLED #3051: Scrolling Text runs correctly only if serpentine is selected

josteir, 2023-01-24:
### What happened?

I want to start saying that this last update with 2D configuration is great! really appreciate that!

So, I've been playing around with pixel art using a matrix 16x16 and works great. I've tons of arts randomly showing in the matrix non-serpentine just fine, because now we don't have to map that anymore.

Now, I'd like to use the Scroll Text effect as well, but unfortunately this effect just shows correctly IF I select the "serpentine" check box in 2D Configuration.
Most of effects I tried are fine, but the Scroll Text one doesn't seem to respond correctly if "serpentine" check box is unchecked.
If I check it, Scroll Text runs perfectly, but then all my pixel arts get scrambled, because they were drawn not thinking in serpentine since we have all those options in 2D configuration now.

My understanding, since I tested a few more effects, is that the effect should follow/respect the options set in the 2D configuration page regarding the serpentine check box.
So, I'm not sure if that is a bug or if I missing something else.

That being said, I'd like to confirm, as a workaround, if it's possible to send a map thru jsonAPI along with the segment and colors. because I'd have to add this mapping to the json I'm sending when changing from pixel art to scrolling text or vice verse,
which would be easier than redo all the pixel arts as serpentine.

thank you!

### To Reproduce Bug

* Set 2D Configuration to "2D Matrix"
* Leave Serpentine check box UNchecked
* run Scrolling Text effect

### Expected Behavior

My understanding, since I tested a few more effects, is that the Scrolling Text effect should follow/respect the options set in the 2D configuration page regarding the serpentine check box.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-24:
Serpentine checkmark tells WLED **how actual LEDs are wired** (how to build internal structures for 2D).
Your external software needs to be updated to take into account that 2D matrix *as exposed by WLED* runs in "non-serpentine" fashion.

This is documented in [KB](https://kno.wled.ge/interfaces/json-api/#per-segment-individual-led-control).

closed 2023-01-24 by blazoncek

josteir, 2023-01-24:
For the pixel arts I mentioned I using CURL to send the info.
and for the Scrolling Text I'm not using any external software. I'm simply selecting "Scrolling Text" effect.
If the serpentine is not checked in the 2d Configuration, it doesn't show correctly. It is like the effect will always run in serpentine. As my understanding, the effect should follow what I've configured in 2D Configuration.