wled/WLED #3053: Improve transition smoothness

jclsn, 2023-01-26:
### What happened?

When you send WLED a command to change the brightness with a [transition time value from e.g. Home Assistant](https://www.home-assistant.io/integrations/light#service-lightturn_on), the results are not great. While this works well on e.g. a Philips Hue of course, the transition on the WLED server is quite flickery and choppy. This behavior even gets worse when you try to repeat transitions in a do-while loop.

### To Reproduce Bug

Try to trigger a light service turn on command changing the brightness to WLED including a transition. Also try doing this in a do-while loop with an automation. The behavior gets even more unrealiable.

### Expected Behavior

WLED should smoothly dim the lights in the given transition time.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14.0-b1

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-26:
Have you tried sending JSON via POST?

jclsn, 2023-01-28:
@blazoncek What do you mean?

blazoncek, 2023-01-28:
[KB](https://kno.wled.ge/interfaces/json-api/)

jclsn, 2023-01-29:
@blazoncek Well, I want to use HA automations. I think HA already uses the JSON. 

But I found the issue. The new 0.14.1b was the main cause. It also works better when you align WLEDs transition time with the transition time in the HA automation. This has a minimum of one second and when I increased the WLED transition from 700ms to 1000ms it worked much better. 

I also reduced the current per LED to 50mA and will solder a better power supply to it and not drive the 36 LEDs with the pin of the ESP8266

closed 2023-01-29 by jclsn