wled/WLED #3054: Issue with WLED + hyperion + Home Assistant combo

Serios, 2023-01-27:
### What happened?

Hi, I'm using WLED with cheap SP511E (ESP8285) controller for ambilight with combination with Hyperion.ng which is working great. WLED HA integration also is working great. Now tried to step up the game a bit - When event is detected in HA, I want some preset to play, then return WLED to previous state (very similar to what hyperion do) . For example, play specific light preset when someone is at front door etc. etc. This actually is working fine... 

During this event the WLED might be on or off, doesn't matter. 
If is off, turn it on, play the preset, then return it to off state (this is working, no problem there). 
If is on (hyperion is sending some data), enable live override, play the preset then return it to hyperion feed, this is working too, except the WLED is turned off just a second after the feed is resumed...

What the issue is that when WLED is OFF and is started by Hyperion feed, the state never changes to ON nor in HA nor in WLED web panel, although the leds powers up and start receiving data from Hiperion, thus later when HA is returning WLED device to previous state in the example above, the leds powers off at the end of the automation sequence since this was their original state, which is wrong. 

### To Reproduce Bug

Power WLED trough Hyperion feed and check the state of the device in HA. The state doesn't change to ON. If you power it manualy trough WLED web panel the state is changed to ON. 

### Expected Behavior

To have On/Off state properly indicated

### Install Method

Self-Compiled

### What version of WLED?

WLED 0.13.3

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-27:
Hyperion uses *realtime* data, HA uses WLED state.
Both are incompatible.

Serios, 2023-01-27:
> Hyperion uses _realtime_ data, HA uses WLED state. Both are incompatible.

Yes, but shouldn't be WLED report power ON state, when the Led's are  powered? Or I misunderstand something.

Zerobyte1, 2023-02-02:
I actually did a similar thing today... watch this video..

https://www.youtube.com/watch?v=9BZuC-LA_XI

I did this exactly and it works perfectly.

Serios, 2023-02-03:
> I actually did a similar thing today... watch this video..
> 
> https://www.youtube.com/watch?v=9BZuC-LA_XI
> 
> I did this exactly and it works perfectly.

@Zerobyte1 thank you, but I already have all that's in the video working (logic, triggers, automations, etc...) the problem is since WLED doesn't change it's state to ON when triggered by Hyperion feed, all kind of issues with the automation logic arise. Maybe I should explain my automation logic to bring some clarity to the case...

Automation logic in general goes lie this:

If event arise (sensor change state) start the fallowing sequence...
1. Save the previous state of WLED (I'm using scenes for this)
2. Turn wled.light to ON (since I want the lights to alarm me no matter if I'm watching TV on that moment or not (no Hyperion feed) this make sure led's turn ON if they are turned off at the moment)
3. Set live override to ON
4. Play the preconfigured light animation (preset) from the WLED playlist
5. Set live override to OFF
6. Return WLED to previous state using scene.turn_on

Now if the TV is off (no Hyperion feed) of the start of the automation sequence - no issue... Led's turn ON, play the preset, then goes OFF again.

If however the TV is on (there is a feed from Hyperion and the led's are turned on by the feed, not manually) 
Led's turn ON (this doesn't matter since they are already working)
Preset plays 
WlED is turned OFF since this is the initial state when the scene is saved (this is killing the ambilight)

blazoncek, 2023-02-03:
The last bit is where your issue is.
External sender (Hyperion) is not aware you switched its live mode off. Sending live mode on would've solved your issue.
WLED does not manipulate live mode internally until it receives on or off command.

Serios, 2023-02-03:
> The last bit is where your issue is. External sender (Hyperion) is not aware you switched its live mode off. Sending live mode on would've solved your issue. WLED does not manipulate live mode internally until it receives on or off command.

Can you elaborate a bit more on this? As I'm understanding I should send live mode on/off to WLED?

blazoncek, 2023-02-03:
WLED requires external device to set it into live mode.
Some protocols like DDP, E1.31 and ArtNet will do it automatically until the feed ceases or you manually cancel it.
Hyperion switches WLED into raw live mode at startup and switches it off upon exit. If you cancel live mode from API call you need to reenable it afterwards.

In any case live mode takes precedence over internal effects.

closed 2024-04-23 by github-actions[bot]