wled/WLED #3055: Palette effect not smooth at low speed since 0.14

PolestarWx, 2023-01-27:
### What happened?

I have tried 0.14-b1 on both my ESP32 (website) and ESP32-c3 (self-compiled) and it works quite well exepted Palette effect that does not run as smooth as before when the speed is low, the led color change abruptly, on 0.13 it was quite smooth event at low speed. It's like the smooth time does not change according to the effecf speed.

### To Reproduce Bug

Install weld 0.14, select Rainbow (or any multiple palette with opposite color) with Palette effect at speed under 30 to notice the bug.

### Expected Behavior

Led color change smooth as in 0.13 for Palette effect.

### Install Method

Binary from WLED.me

### What version of WLED?

0.14-b1

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-27:
Please post comparison video.

Aircoookie, 2023-01-27:
How many LEDs do you have, and what is the FPS as reported in the info tab on the main UI? You can hit the Refresh button after running the Palette effect for at least 5 seconds to get an accurate frames per seconds value. If it is below 30, that would explain the lagging.

blazoncek, 2023-01-27:
Tested, but cannot reproduce.

dosipod, 2023-01-29:
[image]

https://user-images.githubusercontent.com/87589371/215302459-f6263ed6-a068-4955-b602-533ebdbaff0a.mov

blazoncek, 2023-01-29:
As I can see from the image and video above the palette changes and runs smoothly.

PolestarWx, 2023-01-29:
Thank you very for your time about testing this issue, I see you made a new commit to fix this issue, thanks ! 

If you still need video thatshow the problem, I will try to do it soon

blazoncek, 2023-01-29:
@PolestarWx @dosipod please verify if the commit https://github.com/Aircoookie/WLED/commit/c6fd11157a8c68c44e69b4a0157796534e619497 fixes this issue and if so, close it.

PolestarWx, 2023-01-30:
## 0.13
[image]

https://user-images.githubusercontent.com/16896194/215617016-74f40aea-a04f-409d-88fa-5753c4912238.mp4

## 0.14 without commit 
[image]

https://user-images.githubusercontent.com/16896194/215617074-c8f62522-20ff-4d42-afde-bd3c09be6d34.mp4

## 0.14 with your commit 

https://user-images.githubusercontent.com/16896194/215617156-a6bde617-b9d1-476e-9e66-20ab9bce2da5.mp4

## Conclusion
Even better than before !

closed 2023-01-30 by PolestarWx

blazoncek, 2023-01-31:
Unfortunately none of the videos play for me.

PolestarWx, 2023-01-31:
Strange, it's ok on brave and my android browser via

dosipod, 2023-02-01:
I had to convert the videos and if you are okay with the result then better detail that

PolestarWx, 2023-02-01:
it's classic mp4 container with h264 codec :P Don't what going wrong but no problem on my device :p

phoenixtechnam, 2023-05-02:
I have had that issue for a long time on both v0.13 and 0.14. I thought it was an issue with resources on my ESP8266, but my new ESP32 from QuinLED actually does the same thing. As can be seen in the video, the the leading LEDs are not faded but the trailing LEDs have nice and smooth fade-out. This behaviour can be seen in almost all moving effects at very low speeds.

Effect shown: Sinelon Dual, 42 FPS according to live info

https://user-images.githubusercontent.com/29205326/235636566-519ef2f1-6efd-4142-8c4b-2bf22546860c.mov

Kindly assist. I tried all versions including nightlies (0.14-b2) and also MoonModules v0.14.0.b15