wled/WLED #3056: Guide Pixels and LDR Sensor Module

SB-London, 2023-01-27:
**Is your feature request related to a problem? Please describe.**
1. Living in the UK, it is common to have darkness in the house in the middle of the day and have plenty of brightness at 9PM in the summer days. Having stairs module be enabled at a scheduled time isn't feasible.
2. Getting to stairs at night can be difficult unless you turn lights on or have some night night.
3. PIR sensors can be blocked. It is common for people to hang coats and jackets on the stairs bannister. This would block the PIR sensor from working.

**Describe the solution you'd like**
1. Implement LDR module integration so that stairs lights only come on when it is dark regardless of the time of the day.

2.  Implement pilot pixels. First LEDs on the top and bottoms stairs to remain on with random solid colour to enable walking in the dark to approach the stairs. Pilot LEDs can be an option so if a user does not want, they can leave it disabled. (They do help, I'm can vouch for that🤣)

3. Implement Microwave Sensor so that these can be used instead of PIR (typically at the bottom of the stairs). This will go through clothes etc to trigger lights.

**Describe alternatives you've considered**
I have all of these options implemented using an Arduino and custom coding but being able to use wled would be great as it has many lovely effects.

In Arduino, I was able to replace a PIR sensor with a microwave sensor module without any changes to code.

Thanks

blazoncek, 2023-01-28:
Just create a usermod if you have your code written for Arduino.
There are plenty of usermods available as examples, including usermods for LDR and unltrasonic (ToF) sensors.

You can implement 2) with clever ledmaps and segments.

Please check [KB](https://kno.wled.ge) and/or join WLED [forum](https://wled.discourse.group) or Discord for further help and support.

SB-London, 2023-01-28:
Thanks. I'm not much good with code but I'll take a look at it.

blazoncek, 2024-06-19:
Related to #857 implemented in #3490.

closed 2024-06-19 by blazoncek