wled/WLED #3057: wESP32 V7 Ethernet driver not working

curtis8523, 2023-01-28:
### What happened?

The Ethernet is not functioning and has no connection.

### To Reproduce Bug

Load WLED on the most recent version of the wESP32.

### Expected Behavior

When turning on Ethernet in WiFi settings for ethernet to get an IP from the DHCP server and be available to use.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0 -b1

### Which microcontroller/board are you seeing the problem on?

Other

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

curtis8523, 2023-01-28:
Board maker uses the RTL8201 on v7 boards. Previous boards used the LAN8720.

blazoncek, 2023-01-28:
Only LAN8720 is supported by WLED ATM.

bikerglen, 2023-10-14:
Until this change can be implemented, I'd propose changing the "WESP32" label in the Ethernet type pull down menu to "WESP32 (Rev 6 and Lower)."

To get my Rev 7 WESP32 boards to work, I added an experimental "esp32_V4_eth" build environment to my local platformio.ini. This pulls in the ESP32 IDF version 4 which supports the RTL8201 Ethernet PHY. I then changed the PHY type for the WESP32 from ETH_PHY_LAN8720 to ETH_PHY_RTL8201 in network.cpp. This worked.

This is the esp32_V4_eth that I came up with and tested on my WESP32 Rev 7 board. There's some unfortunate copy / paste oversights in there. Also, if WLED moves to V4 of the IDF at some point, I'd add "WESP32 (Rev 6 and Lower)" and "WESP32 (Rev 7)" to the Wi-Fi settings page.

```
[env:esp32_V4_eth]
;; experimental ESP32 env using ESP-IDF V4.4.x
;; Warning: this build environment is not stable!!
;; please erase your device before installing.
board = esp32dev
platform = ${esp32_idf_V4.platform}
platform_packages = ${esp32_idf_V4.platform_packages}
build_unflags = ${common.build_unflags}
build_flags = ${common.build_flags}  ${esp32_idf_V4.build_flags} -D WLED_RELEASE_NAME=ESP32_V4_eth -D RLYPIN=-1 -D WLED_USE_ETHERNET -D BTNPIN=-1
  -D WLED_DISABLE_ESPNOW ;; ESP-NOW requires wifi, may crash with ethernet only
lib_deps = ${esp32_idf_V4.lib_deps}
monitor_filters = esp32_exception_decoder
board_build.partitions = ${esp32_idf_V4.default_partitions}
board_build.f_flash = 80000000L
board_build.flash_mode = dio
```

I'm going to grab another WESP32 pixel controller of mine and run some experiments to try to come up with a smaller / more accurate esp32_V4_eth environment section.

I'm not familiar enough with the WLED development effort to know if a move to V4 is in the works or won't happen. If there's interest in this, I can create a branch, add these changes, and create a pull request.

softhack007, 2023-10-14:
Hi @bikerglen thanks for your research and recommendations 👍 

> I'm not familiar enough with the WLED development effort to know if a move to V4 is in the works or won't happen

Actually, both is true. We have all the changes to use the newer framework (necessary any way for -S3/-S2/-C3), however currently we see only little benefits for esp32. In fact the firmware size of V4 grows by 300KB, compared to the older framework. This would make it hard for people who want a few usermods, as the overall firmware will become too big to fit into 4MB boards with standard partitions.

> If there's interest in this, I can create a branch, add these changes, and create a pull request.

The modified ethernet driver would be interesting indeed. 

Its necessary to still allow building with the "standard" framework, so any special "V4" code needs to be encapsulated with 
```C++
#if defined(ARDUINO_ARCH_ESP32) && (ESP_IDF_VERSION >= ESP_IDF_VERSION_VAL(4, 4, 0))

// code that relies on features od ESP-IDF 4.4.x

#endif
```

bikerglen, 2023-10-15:
Thanks for the info, Frank.

If I'm following the HTML/Javascript/XML code correctly, I think I can do something in this code in xml.cpp to hide or disable the WESP32 (Rev 7 and higher) option when the ESP IDF version is less than 4:

``` 
    #ifdef WLED_USE_ETHERNET
    // *** add my code to show / enable the WESP32 Rev 7+ pull down option here based on IDF version ***
    sappend('v',SET_F("ETH"),ethernetType);
    #else
    //hide ethernet setting if not compiled in
    oappend(SET_F("document.getElementById('ethd').style.display='none';"));
    #endif
```

I'd start with the WESP32 Rev 7+ drop down menu item in the id=ethd division hidden or disabled, then add some code in xml.cpp that detects a build with ESP32 IDF V4 or greater and does an oappend to show or enable the option. Or vice versa.

I know that's not too specific. It's been a decade since I've done any html or javascript coding but I can figure it out. Just want to make sure I'm on the right path to having html in the wifi settings page that can vary with compile time options.

-Glen

curtis8523, 2023-10-15:
Hey everyone I have been able to recompile with the modifications located here

https://wled.discourse.group/t/wesp32-revision-7-eth-phy-rtl8201-support/6558/3

I have installed it onto one of my WESP32's after compiling and verified that it does connect and can use ethernet.

closed 2023-10-15 by curtis8523

bikerglen, 2023-10-15:
@softhack007 I have this working. I'm going to go read the WLED pull request instructions again and so some more testing. 

If the old esp32_eth environment is selected, the option is not shown in the wifi settings page. 

If the new esp32_V4_eth environment is selected, the option is shown and a WESP32 r7+ board can be selected and will connect to a wired Ethernet network.

I don't have any r6 or earlier boards to test on unfortunately.

The exact language on the wifi settings page for the two options is:

WESP32 (Rev 6 and earlier)
WESP32 (Rev 7 and later)

bikerglen, 2023-10-15:
@softhack007 I'm not quite ready to create the pull request (needs more testing) but if you want to take a look around, here's the branch on my fork: 

https://github.com/bikerglen/WLED/tree/add-wesp32-r7-support

blazoncek, 2023-10-15:
> @softhack007 I'm not quite ready to create the pull request (needs more testing) but if you want to take a look around, here's the branch on my fork:
> 
> https://github.com/bikerglen/WLED/tree/add-wesp32-r7-support

It is much easier for us if you create a PR as it can be checked out locally without cloning your repository. It will also show diff.

bikerglen, 2023-10-15:
@blazoncek PR #3444 created. Really wishing I hadn't created PR #3421 on my main now. :) Learning new stuff every day.