wled/WLED #3058: [BUG] Playlist Repeat Does Not Save

mermelmadness, 2023-01-28:
### What happened?

I built a playlist and set it to repeat once since I only want it to run once when activated.  Every time I save it runs once, but when I want to change something the repeat changes to anything between 2 and 9.  Rebooting does not fix the issue.

### To Reproduce Bug

Make a playlist and set it to run once.  The save and let it run.  After, go back to edit it and see what the repeat is set to.

### Expected Behavior

I would expect the repeat setting to remain whatever it is set to initially.

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b2 2301240

### Which microcontroller/board are you seeing the problem on?

ESP32

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

mermelmadness, 2023-02-02:
This is still an issue.
[image]
[image]

blazoncek, 2023-02-02:
I can see wrong repeat count on 1st save but subsequent saves are ok.
Next time please describe a problem in more detail.

mermelmadness, 2023-02-02:
I am not seeing that behavior.  If I uncheck Repeat Indefinitely and set repeat to 1, then save, then immediately click the playlist dropdown, Repeat Indefinitely is checked again.  Restarting does not fix the issue.

dosipod, 2023-02-06:
Old behavior is playlists repeat will default to 2 even if you try to make an effect run only once
[image]

blazoncek, 2023-02-06:
I am not sure I understand what you are saying, so please open a new issue describe it again in every possible detail with examples what you think is wrong and how you think it should work if the problem still persists.
Please use most recent source code as there were some updates. Do not use precompiled binaries available on https://install.wled.me until a new beta is released.

I have tested the recent code and I am able to save repetition count of 1 or more including indefinite repetitions.

alvindotdev, 2023-02-24:
I'm seeing the same issue as last published. 

1. I created a playlist containing several presets
2. I do not want it to repeat indefinitely - i do not want it to repeat at all. so i uncheck that box and set repeat to 1 time
3. I save
4. Immediately the playlist runs on my led strip without repeating
5. If i now expand the playlist again to see the settings, repeat is set to (2)
6. If I view the api call, i can see repeat:2 in the json
7. If i play the playlist anytime going forward, it will repeat twice

I can't seem to permanently make the playlist stay at repeat 1. Even if i update the API call manually to set repeat:1, then save, it will still do the same steps 4-7 above.

This occurs in both the iphone app and the web browser mode

Setting repeat to (1)
[image]

Save and it is not expanded any longer -- it will run the playlist automatically and not repeat which is GOOD
[image]

Any subsequent play to the playlist will repeat (2) times, and when i expand to edit, (2) will be written in the playlist repeat setting
[image]

[image]

Even if i edit the api command directly to repeat:1...
[image]

It will do the same things described above --run without repeat successfully 1 time , and subsequent runs will repeat:2.

[image]

I just flashed wled 1 day ago on the latest version on a new ESP32

dosipod, 2023-02-24:
This post is closed , might be best copy your input into a new post

degracode, 2024-01-01:
Was a new issue opened for this @alvindotdev? I'm seeing this problem exactly as you describe; I change to 1 repeat, and it reverts back to 2.

0.14.0-b1 (build 2212222)

blazoncek, 2024-01-01:
> 0.14.0-b1 (build 2212222)

Use 0.14.1-b2

degracode, 2024-01-01:
> > 0.14.0-b1 (build 2212222)
> 
> Use 0.14.1-b2 or newer.

Are you confirming that a problem was identified and fixed, or just that I should speculatively give the new version a try?