wled/WLED #3059: Segments where you can specify multiple ranges of pixels not adjacent

baelinc, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
I use xlights for my Christmas display and in Xlights under a model or in the case of wled, segments, you can create submodels. So if I have a coro prop, not all leds that I was in the segment are adjacent to each other. Example, I have a prop that is 200 pixels, but I have an area of pixels that I would like as a separate segment or subsegment... This prop is for Easter and is a bunny. the feet outline would be the following pixels: _1-8,12-60,160-200_ and the ear outlines are _66,69-90,132,134-154_. 

**Describe the solution you'd like**
Would like to have the ability to type in the exact pixel number (66) or range of numbers (134-154) separated by a comma that would create all those typed in as a sub segment or even just a segment. Maybe have it like it is now, but have a check box that says advanced and then allows you to get more specific.

blazoncek, 2023-01-29:
You can achieve similar outcome by using ledmaps.

baelinc, 2023-01-31:
> You can achieve similar outcome by using ledmaps.

Is there more information on this as I havent seen ledmaps on wled.

baelinc, 2023-01-31:
Why was the wontfix label added?

blazoncek, 2023-01-31:
Because you can rearrange LEDs into logical groups using ledmap and then use sequential LEDs in segments.
Using a single LED to be part of different segments is possible by virtually extending LED strip length and assigning same led on different places in ledmap.
It may not look best as assigning single LED to several segments (overlapping) does not work well in all circumstances.

baelinc, 2023-01-31:
Do you have anything, a website or youtube video, that goes over this?

blazoncek, 2023-01-31:
[KB](https://kno.wled.ge)

closed 2023-03-28 by blazoncek