wled/WLED #3061: gif support

tarbear123, 2023-01-29:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
i'd like to have gif support

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Thank you for your ideas for making WLED better!

dosipod, 2023-01-30:
Pics are kinda there so baby steps .
https://github.com/werkstrom/WLED-PixelArtConverter

blazoncek, 2023-01-30:
Included with custom comlile of WLED (`-D WLED_ENABLE_PIXART`)
Animated GIFs will require usermod support.

blazoncek, 2023-02-05:
@embedded-creations made animated GIF routines so if you ask him nicely ... who knows. 😉 
A usermod is possible.

dosipod, 2023-02-08:
I think @scottrbailey  is also doing something for gifs  but could be a PI is needed

closed 2025-01-26 by netmindz