wled/WLED #3063: Awtrixcloud

brxw, 2023-01-29:
Danke für deine schnelle Antwort. Ich schreibe mal kurz auf deutsch, um Missverständnisse vorzubeugen xD
Ich meinte nicht die awtrix Matrix sondern viel mehr die Cloud, wo unzählige pixelarts verfügbar sind. Vielleicht kann man sie so implementieren, dass man sie auf der wled Matrix wiedergeben kann. Mit Home Assistant und esphome geht das schon. Da gibt es ein Projekt zu. Aber frag mich nicht wie das gemacht wird, bin 1 Noob.. ^^ Lg

softhack007, 2023-10-05:
@brxw do you have a link to these pixelarts?

Do you know if the cloud providers have an API for integrating their pixelarts into third party apps?
What are their licensing conditions?

brxw, 2023-10-05:
Dear Frank, great to here from you

I never thought, that someone actually answers to this topic. As I know
there is no limitation on using the cloud but I might be wrong.. but i will
send you the link to the iconcloud/discord and so on. Would be nice to hear
from you if you got it working.
Have a great day :-)

And yes there’s an api

API
https://awtrixdocs.blueforcer.de/#/de-de/api

Homepage
https://awtrix.blueforcer.de/apps.html#

Awtrix-Discord
https://discord.gg/cyBCpdx

GitHub
https://github.com/awtrix

brxw, 2023-10-05:
This should be the icon database

https://developer.lametric.com/icons

brxw, 2023-10-12:
Here’s the right api documentation for the LaMetric icon database

https://lametric-documentation.readthedocs.io/en/latest/reference-docs/lametric-cloud-reference.html

KlausHans, 2024-03-25:
Lametric icons in Wled would be great!

netmindz, 2025-01-10:
Is this actually a bug or should it get moved to 0.16 @softhack007 ?