wled/WLED #3065: wrong time in scrolling text #time after boot

gramels, 2023-01-30:
### What happened?

#time in scrolling text effect as default preset shows the unix 0 time (1970) until changed.

### To Reproduce Bug

setup 2d matrix
use scrolling text
set #time as text 
time does not get updated bases in ntp

### Expected Behavior

after ntp correct time is shown

### Install Method

Binary from WLED.me

### What version of WLED?

WLED 0.14.0-b2

### Which microcontroller/board are you seeing the problem on?

ESP8266

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

blazoncek, 2023-01-30:
Please set up NTP configuration in Time & macro settings.

closed 2023-01-30 by blazoncek

gramels, 2023-01-30:
Thanks, assumed it set, didnt look close enough.